erpalma/throttled #124: Support for 2018 Vaio S11 (VJS112C11T)

MingcongBai, 2019-06-27:
Using `throttled` on master, it seems that undervolting does function on this laptop, but the temperature target change was simply ignored. Per the default configuration, under AC power, the temperature target will be set to 95C, and a quick `rdmsr` does confirm this:

```
root@vjs112c [ ~ ] # rdmsr -f 29:24 -d 0x1a2
5
```

However, s-tui still shows a maximum temperature of 80C under full load (compiling Chromium) - albeit that the frequency has been lifted by ~300MHz thanks to undervolting.

Here below is an output of `lenovo_fix.py --debug` (I'm not sure what other information will be useful but I will be happy to provide them):

```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 18
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[D] cpu platform info: feature ppin cap = 0
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 2
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[D] Undervolt plane CORE - write -105 mV (0xf2800000) - read -105 mV (0xf2800000) - match OK
[D] Undervolt plane GPU - write -85 mV (0xf5200000) - read -85 mV (0xf5200000) - match OK
[D] Undervolt plane CACHE - write -105 mV (0xf2800000) - read -105 mV (0xf2800000) - match OK
[D] Undervolt plane UNCORE - write -85 mV (0xf5200000) - read -85 mV (0xf5200000) - match OK
[D] Undervolt plane ANALOGIO - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[I] Starting main loop.
[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 1
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 1
[D] core 0 thermal status: power limit log = 1
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 0
[D] core 0 thermal status: cross domain limit status = 0
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 23
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 1
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 1
[D] core 1 thermal status: power limit log = 1
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 0
[D] core 1 thermal status: cross domain limit status = 0
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 23
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[... 92 lines cut ...]
[D] core 7 thermal status: thermal limit status = 0
[D] core 7 thermal status: thermal limit log = 0
[D] core 7 thermal status: prochot or forcepr status = 0
[D] core 7 thermal status: prochot or forcepr log = 0
[D] core 7 thermal status: crit temp status = 0
[D] core 7 thermal status: crit temp log = 0
[D] core 7 thermal status: thermal threshold1 status = 0
[D] core 7 thermal status: thermal threshold1 log = 0
[D] core 7 thermal status: thermal threshold2 status = 0
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 1
[D] core 7 thermal status: power limit log = 1
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 0
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 25
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match OK
[D] CONFIG_TDP_CONTROL - write 0x2 - read 0x2 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816000dc8160 - read 0xcc816000dc8160 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816000dc8160 - read 0xcc816000dc8160 - match OK
[D] HWP - write "0x20" - read "0x20" - match OK
```

MingcongBai, 2019-06-27:
I should also note that it is currently power throttling at 15W power draw.

nariox, 2019-06-27:
Can you run using `--monitor`? This should show the reason the CPU is being throttled. Bear in mind that modern embedded controllers have been able to limit the performance of CPUs beyond what can be modified with throttled currently.

MingcongBai, 2019-06-28:
@nariox `--monitor` shows power limit.

nariox, 2019-06-28:
What are your settings for power limits?

My EC forces my CPU to a power limit (whenever the dGPU is active), so you might have a similar situation. It is possible that it might be for the best, as the EC could be doing this to protect the VRMs or other circuitry, but I share your frustration.