erpalma/throttled #125: Thinkpad P1 throttling sometimes!?

omnibusor, 2019-07-04:
I'm using dual boot Thinkpad P1 (Ubuntu 16.04, Windows 10).

While my P1 works great without throttling on Windows 10, it suffers from CPU throttling on Ubuntu 16.04. So I found this github project it works great. My P1 (i7-8750H) can reach all core 3.3GHz while stress testing on windows and ubuntu.

However, on ubuntu, sometimes all six CPU clock is stuck at the base CPU clock which is around 900MHz.
It is resolved if I unplug the power cable from the thinkpad and connect it again.
This behavior is quite annoying. Any clue?
Thank you for great project!

erpalma, 2019-07-04:
Hmm... Thermald, tlp or other power management software installed?

omnibusor, 2019-07-05:
It's vanilla ubuntu 16.04.06, thermald is removed and tlp is not installed... I changed power management settings on BIOS but no luck.

DEvil0000, 2019-07-06:
from the description it could have something to do with the switch from battery to ac mode. maybe throttled was in battery mode which may be configured to do such things (check your config). maybe there is some kind of bug in throttled getting fixed when switching the profile. maybe it is not related to throttled at all.Von meinem Samsung Gerät gesendet.

omnibusor, 2019-07-08:
@DEvil0000 Bingo. I unplug the AC power, discharged the battery to 30%. Then, I connect the AC power and see if the throttling occurs. It does not happen at all...

This happens if the battery is fully charged. It seems that my Thinkpad uses battery power if it is fully charged even if AC power is connected.

I will try to find out how to force AC power to always be used when connecting AC. Thanks

DEvil0000, 2019-07-08:
Is your sysFs path configured correct (depends on the kernel e.g. /sys/class/power_supply/AC*/online)?
what does your sysfs report when its throttled?
what is reported by `upower -i /org/freedesktop/UPower/devices/line_power_AC` when throttled?
what does `upower -i /org/freedesktop/UPower/devices/battery_BAT0` report when throttled?
You could also try method dbus instead of polling - then check dbus path `/org/freedesktop/UPower/devices/line_power_AC` which is quite the same as using `upower` I think.

@erpalma the script does not print power source changes between ac and bat right? or what is active/detected at the moment?

omnibusor, 2019-07-09:
1. cat /sys/class/power_supply/AC*/online always shows 1 when AC connected, but throttling exists.
2. While it is throttling, there is no CPU throttling related messages In /var/log/kern.log.
3. /org does not exist on my machine ( Ubuntu 16.04). Maybe it is Unity-based?

Do I need to upgrade my machine to Ubuntu 18.04?

DEvil0000, 2019-07-09:
3) this is not a FS path and may depend on your setup.

The first point should be the most important.

You can try to get more information when starting the script with `--monitor` or `--debug`.
Most likely there is some other software limiting you just in this case.

omnibusor, 2019-07-10:
sudo ./lenovo_fix.py --monitor or --debug always show [AC] and throttling still exists.

BTW, thanks for helping me out. I will share the result if I handle this shit

erpalma, 2019-07-12:
Kernel version? This is strange: "all six CPU clock is stuck at the base CPU clock". Can you post the output of ` grep -R '' /sys/devices/system/cpu/cpu0/cpufreq/` when the CPU is stuck at 900MHz please?

omnibusor, 2019-07-15:
[image]

omnibusor, 2019-07-15:
As described above, this problem seems to be related to the battery. this throttling occurs if the battery is fully charged.

DEvil0000, 2019-07-15:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/erpalma/throttled/issues/125?email_source=notifications\u0026email_token=AAZQPLRJD6MVB527HCZIRK3P7QPZTA5CNFSM4H5X4R72YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZ44HNQ#issuecomment-511296438",
> "url": "https://github.com/erpalma/throttled/issues/125?email_source=notifications\u0026email_token=AAZQPLRJD6MVB527HCZIRK3P7QPZTA5CNFSM4H5X4R72YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZ44HNQ#issuecomment-511296438",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]​

your cpu is idle? whats your load on it? fo you generate load? how do you do generate load?stress -c 12? do you expect it running full power also in idle?

erpalma, 2019-07-15:
Your CPU is > 50% in halt state! There is something that is forcing your CPU in C1, do you have `intel_powerclamp` module loaded?

DEvil0000, 2019-07-15:
@erpalma there are registers holding a table with max power or max multiplier for full load on x cores. this may also be involved here.
And the C0/C1 confuses me since C0 reads like 99.8% and C1 56%

omnibusor, 2019-07-15:
@erpalma I will check the intel_powerclamp tomorrow.
@DEvil0000 the machine can reach all core 3.3GHz under some condition. First, if the battery is not fully charged and the AC adapter is connected. Second, if I unplug the AC adapter and plug it again, the machine can reach all core 3.3GHz for a while, but it is throttling after 2 or 3 minutes.

omnibusor, 2019-07-16:
found a solution in the following link. 
```
https://forums.lenovo.com/t5/ThinkPad-P-and-W-Series-Mobile/Thinkpad-P1-plugged-in-not-charging/td-p/4370761
```
```
 the representative instructed me to shutdown the Thinkpad, turn it upside down and look for a small pin-hole near the middle of the base.  I then used a paper clip, pressed it into this hole and held it for 30 seconds.
```
No throttling at all, it completely disappears and lenovo_fix program works great, the machine can reach all core 3.4GHz!!

erpalma, 2019-07-16:
So basically you just reset the bios?

omnibusor, 2019-07-16:
I think this *hole* is related with some reset stuff, but not sure it reset the bios.
However, reset the bios configuration (i.e., load default configuration in the bios), does not resolve the problem.

erpalma, 2019-07-16:
> However, reset the bios configuration (i.e., load default configuration in the bios), did not resolve the problem.

What? Really? So I guess that the *hole* is also resetting some stuff related to the EC, nice to know!