erpalma/throttled #127: Needs explanation of how exactly to tell when it's working.

abbreviatedman, 2019-07-13:
Or maybe I just need that explanation. :  )

But it seems to e it's not terribly explicit what `--monitor` or `s-tui` will say when `throttled` is working properly or not.

Would be helpful to know the benchmarks we're usually getting throttled at and the marks we're able to hit afterwards, not only for general knowledge but for the purposes of showing what difference is being made and even diagnosing if `throttled` is currently on or not.

I'd be willing to work on the explanation aspect of it, once I get it myself. :  )

DEvil0000, 2019-07-15:
I will try to explain it as briefly as possible and not too technical:
throttled tries to rise the software limits set on the CPU usage to unlock its maximal performance.
so in case your CPU has some work to do it should get it done faster and throttle later.
how this exactly looks in your case depends on your CPU and other software/hardware.

you should know your CPU core count, max frequency and max Watt TDP.
do `lscpu` and maybe look the CPU details up on the intel website (google will help finding it).

stressing the CPU with workload can be done with `stress -c <cores>` where <cores> should be replaced by the number of cores you have (including hyperthreaded cores). As long as stress runs it will add CPU load to those x cores.

so a first easy test would be stressing the CPU with workload without throttled running (when it was running before and was just stopped some changes may still be active - but don't worry).
while stress is running you should have a look at (for example) `s-tui` now. have a look at all 4 graphs over a period of lets say 5minutes.

now you can install, configure and start throttled and repeat the steps from above. you should see a clear difference in the graphs. In case you did see higher power now it works.

what you should expect to see in the graphs:
while your CPU is idle (no stress) you should see some minimal utilization, some low temperature (which is 50°C to 60°C for me), and some low Watt power (about 9W for me), and maybe also a lower frequency but this may not be the case depending on a lot of things. when you start stressing the CPU you should see a quick raise in temperature, a high utilization, a high power in Watt, a higher cpu frequency (turbo) if your CPU has temperature headroom and the ability and some other factors for it.
depending on cooling/temperature rise you would expect the turbo to drop quickly (after seconds maybe) or not even kick in since we stress all the cores (turbo is mostly for loads on 1 or 2 cores - you can test this with e.g. `stress -c 1` or `stress -c 2`). when your CPU temperature hits your limit (100°C minus 3°C to 5°C so about 95°C for modern CPUs) it starts slowing down (frequency wise). also power in watt should go back when the CPU tries to keep the load by holding the temperature. It is normal that multicore loads lower the frequency of the CPU to hold the temperature target. It is quite normal that the CPU hits its max temp (see intel spec) and gets throttled because of that - in case you have good cooling this may not be the limiting factor. The graphs should be quite linear after soem minues - if not there is something else messing with the CPU power settings.

So you actually want to see one things here: a power (in watt) under load as high as possible and close to the max TDP watts of your CPU (see intel spec for the CPU).

@erpalma wiki? readme?

LinuxOnTheDesktop, 2019-07-28:
@abbreviatedman 
There is a command line (well, 'ncurses') program called `s-tui` (which is on github together with installation instructions). That program shows the percentage of performance lost due to the throttling. I think that *this* github repository mentions `s-tui`. So you can use that program to check how well the fix works..