erpalma/throttled #128: Throttle fix seem not to work at boot (but works after boot)

LinuxOnTheDesktop, 2019-07-28:
I find that the throttle fix works well - I'm grateful. You should not have had to do this work (and now you have work maintaining it!) Now for the problem I have: at least of late, it seems not to work at boot. For, at boot, I see lots of messages about throttling:

[image]

**My PC**

```
OS: Linux Mint 19.1 Tessa x86_64 
Host: 20KHCTO1WW ThinkPad X1 Carbon 6th 
Kernel: 5.2.4-050204-generic 
Shell: bash 4.4.20 
DE: Cinnamon 4.2.3  
CPU: Intel i7-8550U (8) @ 4.000GHz 
GPU: Intel UHD Graphics 620 
Memory: 2573MiB / 15927MiB 
Disk (/dev/nvme0n1p4): 17G / 40G (46%) 
Disk (/dev/nvme0n1p2): 186G / 405G (49%) 
Disk (udev): 0 / 7.8G (0%) 
```

**My version of the fix**

The latest git version.

**My config file for the fix**
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
# To reload this config file:
# sudo systemctl restart lenovo_fix.service
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Default: 30.
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Default: 85.
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Default: 5.
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
# As the script came, it was 95. ** THIS IS THE ONE TO CHANGE. ** >= 93 (or lower?) is too noisy?
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Was: False. 'True' may cause throttling on boot.
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: 40
# Was 30. Now 40.
# Integrated GPU voltage offset (mV)
GPU: 30
# Was 10. Now 30.
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: 40
# 100.
# Integrated GPU voltage offset (mV)
GPU: 30
# GPU value: 30.
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0
```

nariox, 2019-07-29:
throttled only works after it is started. You could try to make it run earlier on systemd, but it shouldn't affect you in any major way. If you want to disable the warnings, you could add `loglevel=3` to your boot parameters

erpalma, 2019-07-29:
Correct. Also notice that the "above threshold" messages can still occur when throttled is running. That message indicates that your CPU is being limited but it is perfectly fine since at some point you will reach `Trip_Temp_C`.

LinuxOnTheDesktop, 2019-07-29:
Thanks you for the responses.

@nariox 

I think you mean 'loglevel=2'. For I am at 3 already. However, I like to see some of what is happening so I will leave the level at 3.

@erpalma and @erpalma

Are you two saying that a bit of throttling at boot is nothing to worry about? Whether you are or not, I wonder why, if the cpu is throttling at boot, the computer's fan does not come on; the throttle seems the first resort, which seems odd.

nariox, 2019-07-29:
Yes, this is just letting you know that your CPU reached the trip temp (sometimes set rather low, like 70C or so). Once throttled kicks in, you should have your limits raised. This doesn't affect your system, except for maybe a slightly slower boot (imperceptibly so, I believe).

LinuxOnTheDesktop, 2019-07-29:
> maybe a slightly slower boot (imperceptibly so, I believe)

Hmm. It depends on the size of the throttle (and I suppose whether CPU speed is a bottleneck) but also just logging and printing the messages will take a little time (though I suppose not much and the display can be prevented, at the cost of not printing any warnings).

I'd still like to know why the fan does not come on.

I am interested too in any safe way of starting the throttle fix earlier in boot.

erpalma, 2019-07-30:
> I'd still like to know why the fan does not come on.

I suggest you to re-paste the CPU with high quality thermal compound.

LinuxOnTheDesktop, 2019-07-30:
I am considering that. But that does not answer my question. The question was not, how to stop it? The question was, why does it happen?

erpalma, 2019-07-30:
Low quality thermal compound has low thermal conductivity, thus the cpu temp can grow rapidly because heat is not transferred fast enough to the heatsink. So, in particular during power on, the rise in heatsink temperature does not follow the rise in the cpu temperature but there is some "inertia".

LinuxOnTheDesktop, 2019-07-30:
Right. Thank you. I will close this issue now, unless anyone has any advice on starting the throttle-fix service earlier in the boot process than at present.

erpalma, 2019-07-30:
You can try this service file:

```
[Unit]
Description=Stop Intel throttling 
Before=basic.target
After=local-fs.target sysinit.target
DefaultDependencies=no

[Service]
Type=simple
ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py
Environment=PYTHONUNBUFFERED=1
StandardOutput=syslog
StandardError=syslog

[Install]
WantedBy=basic.target
```

Basically the difference is in the `Before`, `After`, `DefaultDependencies` and `WantedBy` keywords.

nariox, 2019-07-30:
Modern laptops have embedded controllers that can change things like trip temps, max clock, fan speed, etc. Your EC might not turn on the fans unless certain conditions are met (temperature is "high" for x milliseconds, etc). Specially if your cooling solution is limited (like most ultrabooks), you might reach the trip temp without activating the fan.

LinuxOnTheDesktop, 2019-07-30:
@nariox and @erpalma:

You have been tremendously helpful. I appreciate it.

closed 2019-07-30 by LinuxOnTheDesktop