erpalma/throttled #130: lower voltage limit

verfriemelt-dot-org, 2019-08-01:
is there a lower limit to the core offset on 8th gen intel cpus?

i tested all the planes and get computing errors for example beyond -125mv offset for the cache.
but on the cpu core i can go down to -250 without any issues. after i reached -350 i was doubting the values itself and just hit it with -500mv offset of the core which should yield a systemcrash but nothing changed.

the temperatures show quite a big difference between 0 and -250. also the turbo frequency rise from 2.4ghz to ~2.9ghz on all cores under prime95 load.

but between -250 and -500 there is no difference, which seems to tell, that lower values are not possible?

i remember back in the days of a pentium m there was also a minimum core voltage limit of around 0.7volt in idle states - same for current cpus?

verfriemelt-dot-org, 2019-08-01:
```
[AC] Thermal: LIM - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 871 mV - Package: 32.5 W - Graphics: 0.1 W - DRAM: 1.8 W
[AC] Thermal: LIM - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 852 mV - Package: 32.4 W - Graphics: 0.2 W - DRAM: 2.0 W
[AC] Thermal: LIM - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 870 mV - Package: 32.2 W - Graphics: 0.1 W - DRAM: 1.7 W
[AC] Thermal: LIM - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 878 mV - Package: 32.1 W - Graphics: 0.1 W - DRAM: 1.8 W
[AC] Thermal: LIM - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 873 mV - Package: 32.1 W - Graphics: 0.1 W - DRAM: 1.8 W
```

at load with `mprime -t`

erpalma, 2019-08-01:
Ok... I know that there *should* be a connection between voltage planes. For instance CPU and CACHE should be bound to the same value or one of the two cannot be lower than the other. I have never investigated further into this, but I guess I should do it now. Do you see `VCore` value getting lower when you lower CORE voltage below CACHE (e.g. CORE -150, CACHE -100)?

verfriemelt-dot-org, 2019-08-01:
the two seems to be coupled together, if i leave the core at -125 and change the cache to 0 the core voltage goes up.

verfriemelt-dot-org, 2019-08-01:
if i lower it to 150 at cache and 150 at core, i will too get lower values for vcore, but the system becomes instable and freezes :(

verfriemelt-dot-org, 2019-08-02:
there is more weirdness involved.
with the -125mV offset i run with 44watts tpd into the thermal limit at 95C at around 32watt of package power.

after a cold boot, im running always into a 25watt powerlimit with the same steps taken.
after standby / resume i will get the high tpd setting working and will arive at a thermal limit again at around 3.1ghz allcore turbo and 93-95C.

so the configured tdp works only after standby

erpalma, 2019-08-03:
The cold boot thing is quite strange. Have you tried to manually restart throttled service or running it with `--monitor` after a cold boot?

verfriemelt-dot-org, 2019-08-03:
yes, nothing worjed besides standby resume

erpalma, 2019-08-05:
That's strange because I have the same laptop/cpu. Do you have the latest bios? What is your distro? Do you have any non standard bios settings?

verfriemelt-dot-org, 2019-08-05:
debian sid, latest bios via fwupd and no, nothing to fancy in the bios.

erpalma, 2019-08-05:
What about thunderbolt bios assist?

verfriemelt-dot-org, 2019-08-05:
i have to check, but i cannot recall , that i changed any of those. only disable the wwan and switched diagnostics boot. that it.

but i can doublecheck when i reboot the next time.

verfriemelt-dot-org, 2019-08-08:
i noticed some more strange things.
after resume the intel pstate driver scales my cpu always at full turbo. no chance to get it down besides disabling turboboost altogether.

after reboot, pstate works normal.

erpalma, 2019-08-08:
I would suggest you to test a different distro... I mean, pstate issues seems more related to kernel than EC/BIOS. Do you have tlp or tools like that installed?

verfriemelt-dot-org, 2019-08-08:
i have tlp installed
.

erpalma, 2019-08-08:
Maybe some of your issues are related to tlp? Bad config, bugs? I don't know. Actually I'm using tlp too

verfriemelt-dot-org, 2019-08-16:
i disabled tlp, no difference. at the current state, all that performance throttling stuff is a little buggy as of right now.

after standby my cpu clocks are for example locked at full turbo. regardless of powersave or not. i have to figure that stuff out before i can continue here.

erpalma, 2019-08-19:
I would suggest you to try with a different/clean distro. I guess you could just boot from USB.

haarp, 2020-01-20:
Yes, both Core and Cache sit on the same plane and the higher of two determines the minimal voltage for both. The 36C3 Plundervolt presentation also mentioned this effect.

Furthermore, GPU and Uncore/SystemAgent also appear to sit on a shared plane, the same applies here. [unclewebb confirms](http://forum.notebookreview.com/threads/throttlestop-help-undervolt-not-applying.819364/#post-10750181).

It would probably be a very good idea to mention this in the sample config file. Otherwise a lot of time and work can be wasted doing bogus stability testing.

verfriemelt-dot-org, 2020-01-20:
thank you for clarification!
in the meantime i swapped ssds and took the chance to reinstall debian. all the issues are gone now!