erpalma/throttled #133: Undervolting doesn't work X1C7 I7 8665U

NitroBAY, 2019-08-17:
Hello,
undervolt doesn't change anything to my CPU (I see it with lenvoo_fix.py --monitor and by putting crazy undervolting values like -230mv which doesn't affect anything, intel_undervolt was not working either)
X1C7 I7 8665U arch linux :
lenovo_fix.py --debug : 
```
[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 0
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 0
[D] core 0 thermal status: power limit log = 0
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 1
[D] core 0 thermal status: cross domain limit status = 0
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 41
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 0
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 0
[D] core 1 thermal status: power limit log = 0
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 1
[D] core 1 thermal status: cross domain limit status = 0
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 41
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[D] core 2 thermal status: prochot or forcepr log = 0
[D] core 2 thermal status: crit temp status = 0
[D] core 2 thermal status: crit temp log = 0
[D] core 2 thermal status: thermal threshold1 status = 0
[D] core 2 thermal status: thermal threshold1 log = 0
[D] core 2 thermal status: thermal threshold2 status = 0
[D] core 2 thermal status: thermal threshold2 log = 0
[D] core 2 thermal status: power limit status = 0
[D] core 2 thermal status: power limit log = 0
[D] core 2 thermal status: current limit status = 0
[D] core 2 thermal status: current limit log = 1
[D] core 2 thermal status: cross domain limit status = 0
[D] core 2 thermal status: cross domain limit log = 0
[D] core 2 thermal status: cpu temp = 40
[D] core 2 thermal status: temp resolution = 1
[D] core 2 thermal status: reading valid = 1
[D] core 3 thermal status: thermal limit status = 0
[D] core 3 thermal status: thermal limit log = 0
[D] core 3 thermal status: prochot or forcepr status = 0
[D] core 3 thermal status: prochot or forcepr log = 0
[D] core 3 thermal status: crit temp status = 0
[D] core 3 thermal status: crit temp log = 0
[D] core 3 thermal status: thermal threshold1 status = 0
[D] core 3 thermal status: thermal threshold1 log = 0
[... 155 lines cut ...]
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# [ICCMAX.AC]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

# [ICCMAX.BATTERY]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 
```

ilan@ilan-thinkpad ~> uname -r

`5.2.8-arch1-1-ARCH`

erpalma, 2019-08-19:
The undervolt issue is related to #130. Try with the same value for both CORE and CACHE.

NitroBAY, 2019-08-19:
Oh really ? Thank you I was absolutely not aware of this limit.
I'll test it later on ?
Maybe you should (or I may do it and PR) either mentions it in the doc or update the script for the two are linked ?

erpalma, 2019-08-19:
Yep I know, you are right!

ratcashdev, 2020-08-01:
@erpalma what exactly is `PL2_Duration_S: 0.002` in terms of duration? How many seconds?