erpalma/throttled #134: Confirmed working on Asus ZenBook UX430UNR

precupstefan, 2019-08-20:
Thank you very much for developing such a tool. It really helped me when I was about to give up on Linux because of the throttling issues. So as the title states, it's working on Asus ZenBook UX430UNR.

May I suggest adding different presets for each laptop. I know that I can edit the config file to fit my needs, but it took me a while to find this out. I believe that the inclusion of presets will come in handy for plenty of users.

Thanks again!

erpalma, 2019-08-20:
What do you mean with presets? For instance, undervolt is something that is bound to your specific cpu/system and thus you have to find the setting that works for you. Two identical laptops with the same CPU model can indeed handle undervolt differently due to variations in the CPU binning and/or cooling.

precupstefan, 2019-08-20:
I was referring to the temperature configs. The default temperatures, I believe are quite high for my laptop, so I had to adjust them, but in my opinion it would be nice if there was a config file with recommended temps for my laptop, and upon installing to select that file as config for the service.

Regarding undervolting, I didn't take that in consideration when I suggested this, but (I might be wrong) I haven't seen any reports of CPUs that cant handle a minimum of -75/-100 mV undervoltage. That config might be left as it is or be tweaked at a value that it's safe to assume it won't crash the laptop for most users. I'm no expert at undervolting so I don't know how this issue could be handled.

beudbeud, 2019-09-24:
@precupstefan can you share your config file because i have the same laptop. Thanks

precupstefan, 2019-09-24:
`[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 70
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 77
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: 0
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: 0
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# [ICCMAX.AC]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

# [ICCMAX.BATTERY]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 
`

klausenbusk, 2019-09-29:
Note: On the ASUS Zenbook UX430UNR you don't need to use `throttled`, you can just use the [official intel_rapl driver](https://github.com/erpalma/throttled/pull/143).

`throttled` works perfectly, but I don't want to run unneeded software.

precupstefan, 2019-11-02:
@klausenbusk , I tested your sugestion. The values still reset after 5 seconds. Any ideea how make them stop resetting after 5 seconds?

klausenbusk, 2019-11-02:
> @klausenbusk , I tested your sugestion. The values still reset after 5 seconds. Any ideea how make them stop resetting after 5 seconds?

Weird, what values are you changing exactly? I'm only changing `/sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw`.

Could you also post the output of `grep '' /sys/class/dmi/id/bios_*`?

precupstefan, 2019-11-02:
output of bios
`/sys/class/dmi/id/bios_date:04/17/2019
/sys/class/dmi/id/bios_vendor:American Megatrends Inc.
/sys/class/dmi/id/bios_version:UX430UNR.308
`
I'm using all the commands listed in README
`# MSR
# PL1
echo 44000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl/intel-rapl:0/constraint_0_power_limit_uw # 44 watt
echo 28000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl/intel-rapl:0/constraint_0_time_window_us # 28 sec
# PL2
echo 44000000 | /sys/devices/virtual/powercap/intel-rapl/intel-rapl:0/constraint_1_power_limit_uw # 44 watt
echo 2440 | /sys/devices/virtual/powercap/intel-rapl/intel-rapl:0/constraint_1_time_window_us # 0.00244 sec

# MCHBAR
# PL1
echo 44000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw # 44 watt
# ^ Only required change on a ASUS Zenbook UX430UNR
echo 28000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_time_window_us # 28 sec
# PL2
echo 44000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_power_limit_uw # 44 watt
echo 2440 | sudo tee /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_1_time_window_us # 0.00244 sec`

klausenbusk, 2019-11-02:
> output of bios

So we have the same bios version.

> I'm using all the commands listed in README

Could you try running only: `echo 44000000 | sudo tee /sys/devices/virtual/powercap/intel-rapl-mmio/intel-rapl-mmio:0/constraint_0_power_limit_uw # 44 watt` and see if it stick?

precupstefan, 2019-11-03:
The command didn't change anything