erpalma/throttled #136: CONFIG_DEVMEM=y

Utini2000, 2019-08-24:
Hey again,
why does throttled require the kernel to be built with "CONFIG_DEVMEM=y"?
I'm asking because linux-hardened has it set to "CONFIG_DEVMEM=n" which stops throttled from working correctly.

Utini2000, 2019-08-24:
Or maybe there is a way to let throttled work with all the options that do not need "CONFIG_DEVMEM"?

E.g. undervolting and temperature limit should work fine without it.

nariox, 2019-08-30:
You need to access the MSR, so I don't think it should be possible. There is an alternate way to raise the temp limits, but that's about it.

Utini2000, 2019-09-05:
@nariox Undervolt and temp limit works without access to MSR. How ever, currently throttled won't start at all without MSR access.

Maybe you can change the behaviour of throttled so that it will start and all functions that do not rely on MSR will work?

nariox, 2019-09-05:
Please check functions undervolt (line 281), set_icc_max (327), set_hwp (299), etc on https://github.com/erpalma/throttled/blob/master/lenovo_fix.py. They all require writemsr. It is a similar situation with secure boot.

Unless your vendor/bios provides a different way to undervolt or to change HWP/PLs, you are stuck with the Intel DPTF policy approach (only temp limits AFAIK). You could potentially still use monitor mode, but you might find that another tool like s-tui does the monitoring better (although I usually use both to see the VCORE and other values, but since you might not be able to change it, I'm not sure if it'd be useful).

Utini2000, 2019-09-05:
Currently I use intel-undervolt which seems to.work fine for undervolting + temp limit without using MSR. Or is it a fake? Or is throttled simply using a different approach?

https://github.com/kitsunyan/intel-undervolt

nariox, 2019-09-05:
Run `# intel-undervolt read` to check if the undervolt has been applied. But reading `undervolt.c` shows that it also accesses the MSR.

Utini2000, 2019-09-05:
It tells me that the undervolt has been applied.  
I think MSR is only needed to change TPD?

erpalma, 2019-09-06:
Actually if you have write access to the MSR but no `/dev/mem` you could still use the tool. I will change the code to issue a warning rather than an error.

closed 2019-09-06 by erpalma

Utini2000, 2019-09-06:
@erpalma I think the problem is that I jave no access to MSR but access to the rest?

Any then eversthing except TDP change should work (undervolt + temp limit)?

erpalma, 2019-09-06:
The issue was about `/dev/mem` right? It is very easy to check if you have write access to MSR, just try manually with `wrmsr`. Without `/dev/mem` access even TDP is working, at least to some extent. Depending on the laptop you might also need to override the TDP in a MCH BAR register, thus via `/dev/mem`.

Utini2000, 2019-09-06:
@erpalma
Yes you are right, /dev/mem is the issue as it is disabled by linux-hardened kernel. 
See: https://github.com/anthraxx/linux-hardened/issues/11#issuecomment-524631631

It would be very very awesome to still be able to use throttled for undervolting and changing temp limit and even more awesome to also be ablw to change TDP on my Thinkpad X1E without /dev/mem !!! :)

erpalma, 2019-09-06:
You can change the TDP, but I don't think it will work since you also need to override the TDP in the MCH BAR register. Feedback are welcome of course!

Utini2000, 2019-09-06:
@erpalma well I would already be happy to use throttled for undervolting + temp-limit only :)

erpalma, 2019-09-09:
Well, now you can. Let me know if it works as intended

Utini2000, 2019-09-09:
Mhh it the arch repo has not been updated yet. 
I will look into how to install directly from git (shouldn't be too hard from what I remember).

commit c89c1247 (2019-09-06): Relaxed checks for /dev/mem access. Fix #136