erpalma/throttled #137: Lenovo (at least) working on the throttling issue

cm-cc, 2019-08-30:
I just noticed that Lenovo seems to be working on the issue, according to this forum post

https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4513821/highlight/true#M13563

erpalma, 2019-08-31:
Wow this is nice! It seems that we have made a lot of noise ;)

cm-cc, 2019-09-24:
Just to let you know that I "pushed" a little bit lenovo (msg 99 in the above lenovo forum), and today they posted msg 103, with a document which is too technical for me but could be of interest for people here...

kleinph, 2019-09-26:
The document of which @cm-cc is speaking is linked here: https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4534535/highlight/true#M13642

Basically it says that there is a "lap detection" which ensures that the device is not too hot when used on the lap. This lap detection throttling somehow involves closed source Intel stuff which Linux can't use. So there will be an update where this throttling will be done directly in firmware and not via the OS.

According to the [forum](https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4534535/highlight/true#M13642) there is already an update for X1C7 and I can see an update for my T480s (but did not apply it so far).

cm-cc, 2019-09-26:
thanks @kleinph for the clarification. I also own a T480s. If you refer to bios update 1.34 dated Sept 18, it doesn't contain the throttling fix yet. As I understand from lenovo forum, T480s is in the pipeline but they do not say any release date.

vrossum, 2019-10-03:
I installed BIOS 1.26 (30/8/19) on my T480.
This might already contain the fix, because after 
systemctl stop lenovo_fix.service
performance remains high (41W, 95C on multi-threaded CPU task). 
That said, I'm not sure whether  this is the right way to test.

cm-cc, 2019-10-03:
@vrossum I suspect the fix is not implemented yet in 1.26 (it's "old", afaik Lenovo released the fix only for X1 Carbon Gen 7 and T590 up to now).
To test, I think you must first disable lenovo_fix.service (systemctl disable lenovo_fix), **reboot** and check e.g. with s-tui power/cpu speed under load. If it's not fixed, you can reenable and restart lenovo_fix  (systemctl enable lenovo_fix && systemctl start lenovo_fix).

Hugal31, 2020-03-23:
Original link seems dead, here is the new link : https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/td-p/4028489