erpalma/throttled #138: BAT / AC not detected and cannot apply throttling temp

DadouLapous, 2019-09-01:
Hi,
My first issue is that throttled isn't detecting if i'm using AC/BAT,
so I changed the line 
`Sysfs_Power_Path: /sys/class/power_supply/ADP*/online` in the config file
since `ls /sys/class/power_supply/ ` returns `ADP1  BAT1`, and it works.

My second issue is that I cannot apply the throttling temp (notice the ERR in debug log in the end): 

config file 
```
## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 35
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 40
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 80
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -100
# Integrated GPU voltage offset (mV)
GPU: -75
# CPU cache voltage offset (mV)
CACHE: -75
# System Agent voltage offset (mV)
UNCORE: -75
# Analog I/O voltage offset (mV)
ANALOGIO: 0
```
Debug log :
```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 23
[D] cpu platform info: maximum efficiency ratio = 8
[D] cpu platform info: minimum operating ratio = 8
[D] cpu platform info: feature ppin cap = 0
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 1
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 0
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 0
[D] core 0 thermal status: power limit log = 0
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 0
[D] core 0 thermal status: cross domain limit status = 0
[D] Undervolt plane CORE - write -100 mV (0xf3400000) - read -100 mV (0xf3400000) - match OK
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 46
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 0
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 0
[D] core 1 thermal status: power limit log = 0
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 0
[D] core 1 thermal status: cross domain limit status = 0
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 55
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[D] core 2 thermal status: prochot or forcepr log = 0
[D] core 2 thermal status: crit temp status = 0
[D] core 2 thermal status: crit temp log = 0
[D] core 2 thermal status: thermal threshold1 status = 0
[D] core 2 thermal status: thermal threshold1 log = 0
[D] core 2 thermal status: thermal threshold2 status = 0
[D] core 2 thermal status: thermal threshold2 log = 0
[D] core 2 thermal status: power limit status = 0
[D] core 2 thermal status: power limit log = 0
[D] core 2 thermal status: current limit status = 0
[D] core 2 thermal status: current limit log = 0
[D] core 2 thermal status: cross domain limit status = 0
[D] core 2 thermal status: cross domain limit log = 0
[D] core 2 thermal status: cpu temp = 52
[D] core 2 thermal status: temp resolution = 1
[D] core 2 thermal status: reading valid = 1
[D] core 3 thermal status: thermal limit status = 0
[D] core 3 thermal status: thermal limit log = 0
[D] core 3 thermal status: prochot or forcepr status = 0
[D] core 3 thermal status: prochot or forcepr log = 0
[D] core 3 thermal status: crit temp status = 0
[D] Undervolt plane GPU - write -75 mV (0xf6600000) - read -75 mV (0xf6600000) - match OK
[D] core 3 thermal status: crit temp log = 0
[D] core 3 thermal status: thermal threshold1 status = 0
[D] core 3 thermal status: thermal threshold1 log = 0
[D] core 3 thermal status: thermal threshold2 status = 0
[D] core 3 thermal status: thermal threshold2 log = 0
[D] core 3 thermal status: power limit status = 0
[D] core 3 thermal status: power limit log = 0
[D] core 3 thermal status: current limit status = 0
[D] core 3 thermal status: current limit log = 0
[D] core 3 thermal status: cross domain limit status = 0
[D] core 3 thermal status: cross domain limit log = 0
[D] core 3 thermal status: cpu temp = 56
[D] core 3 thermal status: temp resolution = 1
[D] core 3 thermal status: reading valid = 1
[D] Undervolt plane CACHE - write -75 mV (0xf6600000) - read -75 mV (0xf6600000) - match OK
[D] Undervolt plane UNCORE - write -75 mV (0xf6600000) - read -75 mV (0xf6600000) - match OK
[D] TEMPERATURE_TARGET - write 0x14 - read 0x5 - match ERR
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc814000dc8118 - read 0xcc814000dc8118 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc814000dc8118 - read 0xcc814000dc8118 - match OK
[D] Undervolt plane ANALOGIO - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[I] Starting main loop.
```

I'm using the arch package version, on archlinux, with my laptop MSI GL63-8RD (i5 8300h).
My initial CPU crit temp is 100°C, and in s-tui it reaches around 95°C with throttled enabled (instead of 80 in config).

erpalma, 2019-09-01:
The `Sysfs_Power_Path` is a parameter in the config file for this very reason.

Regarding the problem with MSR 0x1A2 I couldn't find any info about a lock bit, but the documentation is quite fragmented. So I guess your vendor has locked that MSR during boot and thus you can't change that value anymore. If you want to lower the temperature just use `thermald` then, still 95°C seems ok for the manufacturer.

DadouLapous, 2019-09-02:
Ok thanks, i'll check thermald

closed 2019-09-02 by DadouLapous