erpalma/throttled #140: X1G7: Configuration?

fr00x, 2019-09-08:
Thanks for providing this tool. I installed s-tui and stress and see that I can't get over 2400Mhz on battery even though the turbo boost of my i7-8565U should go up to 4600Mhz. I run Ubuntu 19.04.

I installed as per the documentation, but did not change anything at all in /etc/lenovo_fix.conf as I have no idea what I would be doing.

Is there a config somewhere for the i7-8565U?

erpalma, 2019-09-09:
I guess that you are limited by bios or tlp.

confuso, 2019-09-22:
For me it also doesn't work well on X1C7, but it's looks more like it is easily overheating. If I have a core on 100%, within seconds the temperature rises to 100°C - sometimes even the fan doesn't have enough time to spin up. Is it supposed to be like that? Without throttled, it looks more reasonable and limiting itself at ~85°C.

fr00x, 2019-09-22:
My system received a firmware update for the embedded controller (via Ubuntu, mentioning performance fixes for us Linux users) and after I applied that fw update, I have s-tui in stress mode showing CPU frequency values way above what I posted two weeks ago.

Looks like Lenovo finally fixed part of their side!

https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN2HHT.firmware

confuso, 2019-09-22:
Thanks a lot. I downloaded, unfortunately "fwupdmgr install ...Lenovo-ThinkPad-X1Carbon7th-X1Yoga4th-ECFirmware-1.10.cab" gave me a "no supported devices found" back. Do I have to take care with using fwupdmgr to update it?