erpalma/throttled #144: T490 & Firmware upgrade

Jacky2k, 2019-10-04:
Hi,
just wanted to let you know that the T490 is also affected with this issue. Your script worked like a charm, but Lenovo released a new firmware for T490, T590, P53s, P43s, which fixes this problem! And they also do provide a update from within Linux.
The update can be found here: https://support.lenovo.com/us/en/downloads/ds539061
It is version 1.54 from Sep., 30, 2019.
Maybe you can include this information in the readme, so that others will get this information. Thanks a lot for the script anyway, it solved the problem for a couple of months for me and I might continue using it for the other features.

HarryWindsor, 2019-10-05:
Hey @Jacky2k I just installed the latest BIOS version. But I still get very similar results:

Running s-tui and `sudo ./lenovo_fix.py --monitor` returns: 
```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[I] Starting main loop.
[D] Undervolt offsets: CORE: -115.00 mV | GPU: -115.00 mV | CACHE: -115.00 mV | UNCORE: -115.00 mV | ANALOGIO: -100.00 mV
[D] IccMax: CORE: 70.00 A | GPU: 31.00 A | CACHE: 6.00 A
[D] Realtime monitoring of throttling causes:

[BATTERY] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 733 mV - Package: 11.2 W - Graphics: 0.0 W - DRAM: 0.5 W
```

s-tui reports 2300MHz and a Perf Lost of 50%, all this at 63C with my fan at 2839rpm.

Once I plug it into AC I get power limit. Performance goes up and down from 2700-3400 MHz. With my Fans at 4000-4500 rpm.

[image]

Do you get the same results?

Jacky2k, 2019-10-06:
Hi,
first off: How do I get this Perf Lost figure? s-tui does not show it.
Then, the most interesting thing is that the command
`# rdmsr -f 29:24 -d 0x1a2`
returns 3 with the new UEFI as it is supposed to be. And as far as I understood the core problem this script is trying to fix is the fact that this value returns 20 (decimal) with the old UEFI version.
The screenshot below was taken while being plugged into AC.

[image]

As you can see the temperature is constant around 97°C which is the maximum value (100-3=97). The package power and clock speed are slowly dropping but that is to be expected once everything heats up.
I have to admit that I had some strange behavior when I tested this for the first time, like oscillations in frequency/temperature/power and sometimes the CPU would only heat to 77°C instead of 97°C, but that was all gone after a reboot and I was not able to reproduce the problem.

Jacky2k, 2019-10-06:
And I get similar results running on battery:
[image]

cm-cc, 2019-10-07:
@Jacky2k  you get the "perf lost" figure by running s-tui under root (or sudo). At least this is what happens on my archlinux machine

amanusk, 2019-10-07:
@cm-cc, @Jacky2k, @HarryWindsor Just want to let you know that the `perf lost` metric in `s-tui` version 0.8.3 is derived as, "The lowest the CPU frequency has dropped, compared to the highest possible frequency available for Boost on all cores".

This metric might be problematic for several reasons, and was thus removed at this point in the newer version (the one showing multiple graphs)

HarryWindsor, 2019-10-07:
@amanusk thank you for explaining the metrics of the `perf lost`. I like the simplicity of s-tui, a great tool.
@Jacky2k yes running `rdmsr -f 29:24 -d 0x1a2` return `3` for me, but I still do not get the same performance as you. I am using pop os, with my mx250 disabled, maybe this causes issues.

[image]

Jacky2k, 2019-10-09:
I guess the problem is that your system does not heat up to the maximum temperature. Did you had a look in the Kernel logs? Is the system complaining of too high temperature?

HarryWindsor, 2019-10-09:
@Jacky2k thank you for your help. After some looking around I found that my problem was that the CPU never used more than ~13W of power. This is because the `scaling_governor` always stayed on `powersave`. Once I switched it to `performance` I got higher temperatures and better speed. I still use throttled for undervolting with -115mV everywhere. The max. temperature of 85C is set by throttled too. 

Thank you for your help, finally I can use the power I payed for.

[image]

erpalma, 2019-10-09:
@HarryWindsor `powersave` but with which driver? Because if you are with `intel_pstate` then it's fine.

HarryWindsor, 2019-10-09:
I just check what driver I am using and you are right, I am using `intel_pstate` but without switching to the `performance` governor I am stuck at ~13W and 2300MHz.

erpalma, 2019-10-10:
What about `energy_performance_preference`?

erpalma, 2019-10-11:
Ok that's fine. Can you try `HWP_Mode: True`?

HarryWindsor, 2019-10-11:
Yes, this gives me the same performance as manually setting `scaling_governor` to `performance` while I am connected to AC.

tomgs, 2020-03-14:
Sidenote: I've encountered the same exact problem with a T490 (S/N start with `N20`), and using `throttled` alone wasn't bringing me over 2.7 GHz (my CPU can hit 4.6). Played around `scaling_governor`, the `throttled` conf file and updated the firmware, to no avail.

However, using the [CPU Power Manager](https://extensions.gnome.org/extension/945/cpu-power-manager/) by @martin31821 100% brings me to the max (source code for the extension is [here](https://github.com/martin31821/cpupower)). Thank you martin!

If I understand correctly from here, most of what the extension is doing is playing around with files in `/sys/devices/system/cpu/intel_pstate`, as suggested earlier in this thread. Just wanted to put it out there for anyone's who's interested.

tomgs, 2020-03-14:
Alright, was happy too fast - it worked fine for a while, but I'm being throttled again now.

PaJaSoft, 2022-03-22:
It might help somebody - my results for sustainal 'power' setting (T590, i7-8665U, fan in 'auto' mode):
`cpupower frequency-set -d 400MHz -u 4800MHz -g performance -r`
`cpupower set -b 0`
+ undervold everything to -115mV

Up to 4 cores (4 real cores in one package - 1 numa) are able to run in ~4.2GHz with temperature ~70-75°C without any throttling event occured and up to 8 cores (with MT) up to ~3.6GHz.

My only limit is caped to 25W sustained power for package-0 (CPU) which I failed to overcome with any setting so far (turbo mode up to 1 minute - depeds on utilization however 44W is too much - ~30W seems appropriate and cool-able).