erpalma/throttled #145: Add F31 builds to the copr repo

BjoernDaase, 2019-10-05:
There are currently no F31 builds in the copr repo. Even though F31 is still in the beta, it would help to get a build up and running and tackle problems early.

godenji, 2019-10-27:
Probably want to file this issue in the Fedora/Copr repo: https://github.com/abn/throttled-rpm

BjoernDaase, 2019-10-27:
@godenji The issue with exact that question was already closed there over a month ago without any reason. So I decided, to open up an issue over here...

godenji, 2019-10-28:
I see, using the undervolt project for this reason, just clone the repo and
you're ready to go (though I'd like to try some of the additional features
in this project).

nsteinmetz, 2019-11-02:
@BjoernDaase look at some forks, there are some f31 packages available - even if it's strange that F31 has not been added yet for the "officials" rpms.