erpalma/throttled #147: Can I use this on the T490s?

StarrKiss, 2019-10-29:
Just what the title says. I mainly want to use it for undervolting (among other things) and would like to know in case it screws up my system.

j0k2r, 2019-10-31:
Same problem here.

I solved the problem by updating the BIOS from 1.55 to [1.59 (31 Oct 2019)](https://pcsupport.lenovo.com/fr/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/driver-list/component?name=BIOS%2FUEFI).

StarrKiss, 2019-11-02:
What problem are you having @j0k2r ? I am using it now on my t490s w/ no problems.

j0k2r, 2019-11-04:
@StarrKiss, I was talking about the BIOS update, I thought it will solve the throttling issues in Linux without using throttled.

boogy, 2019-11-29:
> Same problem here.
> 
> I solved the problem by updating the BIOS from 1.55 to [1.59 (31 Oct 2019)](https://pcsupport.lenovo.com/fr/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/driver-list/component?name=BIOS%2FUEFI).
> 
> Edit:
> I was too enthusiast, the problem persist even with the new BIOS version !

The latest BIOS release is 1.55. Where did you find the 1.59 version ?
I've updated to BIOS version 1.55 and the throttling problem is still present. Can someone else confirm this please ?

j0k2r, 2019-12-02:
@boogy The 1.59 version was released by Lenovo in (31 Oct 2019) and it was deleted 2 days after that because of some bugs (It was not very professional).

The latest BIOS release 1.55 [provides](https://download.lenovo.com/pccbbs/mobiles/n2jur17w.txt): 
* UEFI BIOS 1.55
* **ECP(Embedded Controller Program) 1.13**

Not sure if it will help, but you can try to update the **ECP to 1.15** from: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JHT.firmware