erpalma/throttled #150: support for X1 Carbon 7th i7-10710u

YihanWangAstro, 2019-11-22:
Is it safe to just comment the CPU check in the script to get support for 10710u?

if not, I would really appreciate it if you can add support for 10710u. Thanks in advance!

erpalma, 2019-11-23:
Just try with the `--force` command line option! If everything works fine please report back so I can add Gen10 CPUs to the list.

YihanWangAstro, 2019-11-23:
> Just try with the `--force` command line option! If everything works fine please report back so I can add Gen10 CPUs to the list.

it works! Thank you!

AdnanHodzic, 2019-12-12:
Can confirm that it also works on X1 Carbon 7th gen i7-8565U.

YihanWangAstro, 2019-12-12:
> 
> 
> I followed[ Debian/Ubuntu install instructions](https://github.com/erpalma/throttled#debianubuntu) but I still get throttling on X1 Carbon 7th gen i7-8565U.
> 
> Same thing happens when I run `sudo ./lenovo_fix.py --force`
> 
> Any ideas?

Can I see your `/etc/lenovo_fix.conf`? The default TDP may not be good for 8565U.

AdnanHodzic, 2019-12-13:
Of course:

```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: 0
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: 0
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# [ICCMAX.AC]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

# [ICCMAX.BATTERY]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 
```

YihanWangAstro, 2019-12-13:
> Of course:
>
> ```
> [GENERAL]
> # Enable or disable the script execution
> Enabled: True
> # SYSFS path for checking if the system is running on AC power
> Sysfs_Power_Path: /sys/class/power_supply/AC*/online [...]

Thank you.

First, I recommend to use `sudo systemctl status lenovo_fix.service`  to check if the fix service works properly. This is what I get

```
lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
   Active: active (running) since Thu 2019-12-12 18:52:17 EST; 9min ago
 Main PID: 1023 (python3)
    Tasks: 2 (limit: 18766)
   Memory: 16.1M
      CPU: 399ms
   CGroup: /system.slice/lenovo_fix.service
           └─1023 /opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py --force

Dec 12 18:52:17 M87-Nebula systemd[1]: Started Stop Intel throttling.
Dec 12 18:52:17 M87-Nebula python3[1023]: [I] Loading config file.
Dec 12 18:52:17 M87-Nebula python3[1023]: [I] Starting main loop.
```  

- If it doesn't work well, edit `/etc/systemd/system/lenovo_fix.service`

    add `--force` after `ExecStart`. 
    this is my `lenovo_fix.service`
   ```
   [Unit]
Description=Stop Intel throttling 

[Service]
Type=simple
ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py --force
# Setting PYTHONUNBUFFERED is necessary to see the output of this service in the journal
Environment=PYTHONUNBUFFERED=1
StandardOutput=syslog
StandardError=syslog

[Install]
WantedBy=multi-user.target
   ```
restart the service by using `sudo systemctl restart lenovo_fix.service` and reboot. Check if you still getting throttling.

- If it works well already. I recomend you to lower your TDP in `/etc/lenovo_fix.conf`. Here is my config

```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 15
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 35
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 80
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 20
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 40
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 90
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -85
# Integrated GPU voltage offset (mV)
GPU: -50
# CPU cache voltage offset (mV)
CACHE: -85
# System Agent voltage offset (mV)
UNCORE: -50
# Analog I/O voltage offset (mV)
ANALOGIO: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -85
# Integrated GPU voltage offset (mV)
GPU: -50
# CPU cache voltage offset (mV)
CACHE: -85
# System Agent voltage offset (mV)
UNCORE: -50
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# [ICCMAX.AC]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

# [ICCMAX.BATTERY]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 
```

it works perfect on my i7-10710u.

ulises-castro, 2019-12-31:
Thanks a lot of @YihanWangAstro, it's works perfect on my Thinkpad Carbon Gen 7 (Pop os 19.10 based on Ubuntu 19.10).

commit 88216b11 (2020-02-20): add Comet Lake to the list of supported cpus (fix #150)
closed 2020-02-20 by erpalma