erpalma/throttled #153: Getting error during installation.

shivindera, 2019-11-26:
`Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Requirement already satisfied: configparser==3.8.1 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (3.8.1)
Collecting dbus-python==1.2.8 (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/3f/e7/4edb582d1ffd5ac3c84188deea32e960b5c8c0fe1da56ce70224f85ce542/dbus-python-1.2.8.tar.gz
Collecting PyGObject==3.32.2 (from -r requirements.txt (line 3))
Collecting pycairo>=1.11.1 (from PyGObject==3.32.2->-r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/3c/1a/c0478ecab31baae50fda9956547788afbd0ca563adc52c9b03cab30f17eb/pycairo-1.18.2.tar.gz
Building wheels for collected packages: dbus-python, pycairo
  Running setup.py bdist_wheel for dbus-python ... error
  Complete output from command /opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2bynglmm/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-d4ofh99j --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for dbus-python
  Running setup.py clean for dbus-python
  Running setup.py bdist_wheel for pycairo ... error
  Complete output from command /opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2bynglmm/pycairo/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-syw4xt8i --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for pycairo
  Running setup.py clean for pycairo
Failed to build dbus-python pycairo
Installing collected packages: dbus-python, pycairo, PyGObject
  Running setup.py install for dbus-python ... |^canceled
Operation cancelled by user
`

Laptop: Thinkpad X1 Carbon
Ubuntu: Ubuntu 19.10
Processor: Intel® Core™ i7-8565U CPU

I installed the pre-requisites but still getting the above error

shivindera, 2019-11-26:
Nevermind, fixed it by running `sudo pip install -r requirements.txt`

closed 2019-11-26 by shivindera

erpalma, 2019-11-26:
That wasn't an actual error and you didn't fix it by running pip with sudo. The `Failed building wheel for X` is just because `wheel` is not installed in the virtualenv. If the package does not include a `wheel` in it, then pip tries to build one and then install it. It fails to build the wheel since `wheel` package is not available and fall back to a regular install. This is more of a warning than an error and I guess that this pip behavior will change in the future.

shivindera, 2019-11-26:
Yeah the thing is I had everything installed, I had to modify the install.sh script to make it `sudo pip install -r requirements.txt` instead of just `pip install -r requirements.txt`. And also the service failed to start after the install was successful. I had to modify `lenovo_fix.py` to add 
```
import sys
sys.path.insert(0, "/usr/lib/python3/dist-packages")
```
as it was failing to import dbus, which again was already installed. I really couldn't figure out why it was happening, but in the end it worked. Thanks for your script!! Now I have to figure out why the laptop gets so hot on Ubuntu compared to Windows by just opening apps.