erpalma/throttled #156: Correct PL1 / PL2 values on H series cpu

korommatyi, 2019-12-11:
I use a lenovo x1 extreme gen 2 machine that has the intel i7-9750h CPU. As I understand, the default PL1 / PL2 values `throttled` comes is for U series CPUs. Can you please help me what are the safe values for H series CPUs?

I've done some research and it seems to me that I should set PL1 to 45, PL2 to 80. I would be a great help if you could verify that these are indeed the expected values.

chaoyi, 2019-12-11:
I use lenovo p1 gen2 that has i9-9880h.  I set PL1 to 120, and PL2 to 4095 for no reason.  I am pretty sure my ```throttled``` is running and my machine does not have any negative effect.  BTW, I use -125mv undervolt for AC.

haarp, 2020-01-08:
Power limits are called TDP by Intel. You can look up the long-term limit on https://ark.intel.com. The short-term limit for Intel CPUs is usually 10W more than the long-term one.

On my X1E Gen2, that would be 45/55W. 80W, 120W, or even no limit seems pretty insane to me. The CPU will almost instantly hit its temperature limit at such powers. And even tho this won't harm the it, I'm not so sure how long the power regulators will survive such massive abuse.

I've commented the PL_Duration settings in the config, as I see don't know the stock value either, and there's little reason to mess with those IMO.

chaoyi, 2020-01-08:
I have temperature limit set at 97C.  With good thermal paste and undervolt, my i9-9880h can sustain CPU package power of about 65W.

With throttled, I have very high geekbench score: https://browser.geekbench.com/v5/cpu/624400

roliverio, 2020-01-14:
I've been using -125 on X1 Extreme gen2 with a i9750, however, i'm unsure of the other values, can someone give me some pointers on a good config here?. i'm weary of increasing max temp values.

chaoyi, 2020-01-14:
I don't know about long term effect, but setting a higher PL1 will speed up long running CPU-intensive tasks.  (your CPU will still be thermal throttled, but with a higher sustained power)

Also, I think 97C on AC is safe because that's the value used by Windows if you are running latest BIOS, drivers and set profile to maximum performance.

korommatyi, 2020-01-14:
I think the best you can do is to restore Windows on your machine and check what is the default value for PL1 and PL2.

I was too lazy to do so. :smile: Instead I found [this](https://www.ultrabookreview.com/31168-lenovo-thinkpad-x1-extreme-gen2-review/) review. They have a picture in the "Gaming" section that shows different graphs for CPU and GPU frequency. On that picture, on the right hand side, you can see PL levels. In max performance mode, `PL1=45`, `PL2=75`. I assume these are the factory settings, so I went with those values. I set PL2 duration to 20 seconds (though, I always hit the temperature limit of 95C before 20 seconds even with my -160mV undervolt...).

So far everything works fine and I got nice benchmark results (over 1200/6000 on Geekbench5), but I have no idea what are the long term effects of such settings.

roliverio, 2020-02-21:
I'm using now Debian testing and the throttling issue went away, installing this service in fact causes speed bursts and noticieable throttling whenever there is a high cpu load, so i've disabled it.

braveally, 2020-05-17:
> I think the best you can do is to restore Windows on your machine and check what is the default value for PL1 and PL2.
> 
> I was too lazy to do so.  Instead I found [this](https://www.ultrabookreview.com/31168-lenovo-thinkpad-x1-extreme-gen2-review/) review. They have a picture in the "Gaming" section that shows different graphs for CPU and GPU frequency. On that picture, on the right hand side, you can see PL levels. In max performance mode, `PL1=45`, `PL2=75`. I assume these are the factory settings, so I went with those values. I set PL2 duration to 20 seconds (though, I always hit the temperature limit of 95C before 20 seconds even with my -160mV undervolt...).
> 
> So far everything works fine and I got nice benchmark results (over 1200/6000 on Geekbench5), but I have no idea what are the long term effects of such settings.

How's the system working these days. I was concerned about the amount of throttling of the CPU on the X1. Are you still running stable with no problems since changing the values?

korommatyi, 2020-07-22:
Yep, everything works fine.