erpalma/throttled #163: Add BDPROCHOT

DOGGO20, 2020-01-02:
This value in Throttlestop on Windows 10 helps with power limit throttling on my Dell Inspiron 15 7567 laptop and from my brief testing disabling it also works in linux.
The MSR address is 0x1FC. So for example using `sudo rdmsr 0x1FC` I get an output of `2c005d` To disable BD PROCHOT, I need to change the value of bit 0 from 1 to 0. So the new value of 0x1FC is `2c005c` by doing this my CPU goes from 800mhz to 3.1Ghz on all core load and GPU under load as well. This does increase thermals so others should keep note of this but it will definitely help prevent throttling! cheers!

nariox, 2020-02-17:
Hi @Hyper-KVM, I've implemented the function, see if it works for you. It doesn't seem to do anything on my computer, but I hope it works for you.

Hyper-KVM, 2020-02-20:
haha! excellent! thank you @nariox I shall test it at some point. :)

closed 2020-02-20 by erpalma