erpalma/throttled #165: Can this work on thinkpad e14?

renato145, 2020-01-26:
Hi, I am facing the same issue with a thinkpad e14. Which things will I need to take care in order to more safely try this solution?

nariox, 2020-02-17:
I don't think there is any "real" risks for regular use. The "risks" (and functionality) are about the same as throttlestop. The main "issue" is that you can't use it with secure boot (actually, only some of the functionality does work, but I forgot which ones don't).

It might be a "security" risk, in the sense that the program is writing into the MSR, but as long as you trust the software, it should be fine (this should be true for any program you run on your computer, specially as root).

That said, there are some settings that might cause issues (like removing current limits or BDPROCHOT). Undervolting is safe always to the hardware (but may cause crashes, so careful with your data) and the power limits and temp limits are safe as long as your laptop was designed properly, it won't be an issue.

PS.: I'm assuming you are the same Renato from the Lenovo support thread?

renato145, 2020-03-07:
Yes. I'm from the forum. Trying this now thanks!