erpalma/throttled #167: Lenovo X280?

skluth, 2020-01-28:
I have a Lenovo X280 (ubuntu 18.04) which has the throttling issue. The CPU frequencey is at 400 MHz constantly, even when I run a stress test :(. 

I tried the packge, installed fine, and the monitoring ans debug options don't show any prolems either. 

However, even with throttled the CPU stays at 400 MHz under all circumstances. 

Do you have any idea why this could happen? Should I check something specific?

Thanks for you great work! Stefan

skluth, 2020-01-28:
I forgot, I updated to latest bios (1.36) and ECP (1.14) dated 17 Jan 2020 but that didn't change anything too.

nariox, 2020-03-06:
Do you have any program running that might be interfering with throttled (e.g. thermald, tlp, laptop-mode-tools, powertop, etc)? TLP had a really aggressive policy for battery by default some time ago.

Are you running on battery or AC? You might want to try resetting your BIOS to default settings. Does it also happen on Windows (if you still have it installed).

skluth, 2020-03-06:
It is always throttled, I'll have to try to reset the bios, but for that 
I have to open the laptop, and disconnect the bios battery, according to 
Lenovo. I'll try that in the next few days. No test with Windows as I 
don't have it.

On 2020-03-06 22:00, nariox wrote:
> Do you have any program running that might be interfering with
> throttled (e.g. thermald, tlp, laptop-mode-tools, powertop, etc)? TLP
> had a really aggressive policy for battery by default some time [...]

nariox, 2020-03-07:
The EC will throttle down if any of the temperatures in the board are too high. Any chance a sensor might have been damaged (e.g. liquid damage)?

Some distros have these other programs installed and enabled by default. Might be worth checking.

By the way, how are you measuring the frequency? Are you using s-tui?

muniter, 2020-04-16:
@skluth Have you checked if it's throttled also when plugged in to power?