erpalma/throttled #171: 'Overriding invalid values'

LinuxOnTheDesktop, 2020-02-03:
I have the latest version of your program and it is reporting errors in my config file but I am afraid I cannot find them. So I am inclined to think that either the program is not parsing the config properly or else it could at least give more informative error messages :)

Here are the errors.

```
~ $ cat /var/log/syslog | grep 'Overriding invalid'
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.AC" value in "CORE" voltage plane: 40 -> 0
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.AC" value in "GPU" voltage plane: 30 -> 0
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "CORE" voltage plane: 45 -> 0
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "GPU" voltage plane: 30 -> 0
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "CACHE" voltage plane: 30 -> 0
Feb  3 04:53:42 x1 python3[1357]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "UNCORE" voltage plane: 30 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.AC" value in "CORE" voltage plane: 40 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.AC" value in "GPU" voltage plane: 30 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "CORE" voltage plane: 45 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "GPU" voltage plane: 30 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "CACHE" voltage plane: 30 -> 0
Feb  3 04:59:04 x1 python3[19769]: [!] Overriding invalid "UNDERVOLT.BATTERY" value in "UNCORE" voltage plane: 30 -> 0
```

Here is my config file.

```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
# To reload this config file:
# sudo systemctl restart lenovo_fix.service
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Default: 30.
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Default: 85.
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Default: 5.
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
# As the script came, it was 95. ** THIS IS THE ONE TO CHANGE. ** >= 93 (or lower?) is too noisy?
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Was: False.
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: 40
# 40.
# Default: 0
# Integrated GPU voltage offset (mV)
GPU: 30
# 30.
# Default: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: 45
# Default: 0
# 45. 50 causes VirtualBox problems? 80 seems to. 100: caused a crash.
# Integrated GPU voltage offset (mV)
GPU: 30
# Default: 0
# GPU value: 30.
# CPU cache voltage offset (mV)
CACHE: 30
# Try 30.
# System Agent voltage offset (mV)
UNCORE: 30
# Try 30
# Analog I/O voltage offset (mV)
ANALOGIO: 0
# Try 30?
```

**My system**

* ThinkPad X1CG6
* Linux Mint 19.3 x64 Cinnamon
* 5.3 kernel

nariox, 2020-02-03:
Your undervolt values should be negative. See if it still happens if you add a minus sign in front of all undervolt entries.

LinuxOnTheDesktop, 2020-02-03:
Making the values negative did indeed fix the problem. This will be something that trips many people up, I think. After all, the sections are entitled _under_volt. I suggest that you put something about this as a comment at the top of the default conf file. Thanks for the help.

nariox, 2020-02-03:
The Readme states that the values must be negative (it also has an example). But I see how it could be confusing by only looking at the conf file.
You can check the relevant section of the readme here: https://github.com/erpalma/throttled#configuration

LinuxOnTheDesktop, 2020-02-03:
Ah. Thanks. Still, you might as well stick the point in the conf file as well, no?

commit d5ee6de8 (2020-02-20): mention that all undervolt voltage must be negative in config file (fix #171)
closed 2020-02-20 by erpalma