erpalma/throttled #174: It seems that removing the Intel Management Engine SOLVES Thermal Throttling Issues

CodeDragon5793, 2020-02-06:
Using the me_cleaner tool, I disabled the Intel Management Engine on my ThinkPad T480. Upon immediate reboot I noticed what appeared to be improved thermals. Thus I decided to test it. I removed the throttled daemon from my system completely and ran a five minute stress test using s-tui. This is still preliminary results. I don't know what else will come of this.
[image]

For more information on what I did visit: https://github.com/corna/me_cleaner/issues/3 and https://github.com/corna/me_cleaner/wiki/External-flashing

CodeDragon5793, 2020-02-06:
FYI Hyperthreading on my system is disabled because of Zombieload

CodeDragon5793, 2020-02-06:
Note: My laptop was plugged in during that test.

CodeDragon5793, 2020-02-06:
The Intel ME doesn't stay disabled.

closed 2020-02-06 by CodeDragon5793

asvc, 2020-02-14:
> The Intel ME doesn't stay disabled.

Do you mean that `HAP` bit somehow reverts back?

jonathanpopham, 2020-11-29:
Hello, @CodeDragon5793  can you please clarify if the HAP bit reverts? Was this unsuccessful? I am considering a t480 but I don't want to buy one if ME cannot be disabled.