erpalma/throttled #175: CPU and iGPU clock unit

kikislater, 2020-02-11:
Thank you for this software it helps a lot. 

To solve recent kernel change and lot of battery drain, I have to manually set iGPU clock. It's cool that throttled do it, however, units are not the same : Hz for CPU and MHz for iGPU. It's a bit confusing IMO

What is working in my configuration is the following > 

# Set the min/max frequency available for the scaling governor.
# Possible values depend on your CPU. For available frequencies see
# the output of tlp-stat -p.
# Default: <none>
CPU_SCALING_MIN_FREQ_ON_AC=800000
CPU_SCALING_MAX_FREQ_ON_AC=4000000
CPU_SCALING_MIN_FREQ_ON_BAT=800000
CPU_SCALING_MAX_FREQ_ON_BAT=800000
 

# Set the min/max/turbo frequency for the Intel GPU.
# Possible values depend on your hardware. For available frequencies see
# the output of tlp-stat -g.
# Default: <none>
INTEL_GPU_MIN_FREQ_ON_AC=300
INTEL_GPU_MIN_FREQ_ON_BAT=300
INTEL_GPU_MAX_FREQ_ON_AC=1150
INTEL_GPU_MAX_FREQ_ON_BAT=300
INTEL_GPU_BOOST_FREQ_ON_AC=1
INTEL_GPU_BOOST_FREQ_ON_BAT=0

haarp, 2020-02-14:
Are you confusing throttled with tlp?

kikislater, 2020-02-14:
Oh yes sorry !

closed 2020-02-20 by erpalma