erpalma/throttled #180: Error with PyGObject wheel

ElDmitro, 2020-03-09:
Hello,

While installing fix, some error occure:

>Collecting` configparser==3.5.0
  Downloading configparser-3.5.0.tar.gz (39 kB)
Collecting dbus-python==1.2.8
  Downloading dbus-python-1.2.8.tar.gz (787 kB)
     |████████████████████████████████| 787 kB 2.9 MB/s 
Collecting psutil==5.4.6
  Downloading psutil-5.4.6.tar.gz (418 kB)
     |████████████████████████████████| 418 kB 3.4 MB/s 
Collecting PyGObject==3.28.3
  Downloading PyGObject-3.28.3.tar.gz (698 kB)
     |████████████████████████████████| 698 kB 5.7 MB/s 
Collecting python-periphery==1.1.1
  Downloading python_periphery-1.1.1-py2.py3-none-any.whl (19 kB)
Collecting pycairo>=1.11.1
  Downloading pycairo-1.19.1.tar.gz (205 kB)
     |████████████████████████████████| 205 kB 8.1 MB/s 
Building wheels for collected packages: configparser, dbus-python, psutil, PyGObject, pycairo
  Building wheel for configparser (setup.py) ... done
  Created wheel for configparser: filename=configparser-3.5.0-py3-none-any.whl size=20938 sha256=e552035bb1ca3e4070f7a9c4cacbe6cb6d28e56b1e2deee22d5e7b5f9205ff05
  Stored in directory: /tmp/pip-ephem-wheel-cache-020u07tl/wheels/5f/b6/c5/c6d8ccf999b401c86f31f417db9fe2c7a970e9232847c55a5a
  Building wheel for dbus-python (setup.py) ... done
  Created wheel for dbus-python: filename=dbus_python-1.2.8-cp36-cp36m-linux_x86_64.whl size=367365 sha256=54af0ded8b8a860deb0618e60fb168ef9539e6039d09ffc6d436426b8c27089d
  Stored in directory: /tmp/pip-ephem-wheel-cache-020u07tl/wheels/87/71/42/44755d5b11d648a31e1c2dcfe612737839f99af345b1207453
  Building wheel for psutil (setup.py) ... done
  Created wheel for psutil: filename=psutil-5.4.6-cp36-cp36m-linux_x86_64.whl size=265595 sha256=a05aa71bebfc11823130888629d14ca8a4b75222920195b250a7dfa22565ac1d
  Stored in directory: /tmp/pip-ephem-wheel-cache-020u07tl/wheels/9e/93/1a/07bd1923b508b133b4abeb546e345280a7150a93da4add3f60
  Building wheel for PyGObject (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4ex2257z/PyGObject/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4ex2257z/PyGObject/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-j1hs8jth
       cwd: /tmp/pip-install-4ex2257z/PyGObject/
  Complete output (73 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.6
  creating build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/__init__.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/pygtkcompat.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/generictreemodel.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  creating build/lib.linux-x86_64-3.6/gi
  copying gi/_ossighelper.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/docstring.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/__init__.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/pygtkcompat.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_constants.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_error.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/module.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/types.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/importer.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_option.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_signalhelper.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_propertyhelper.py -> build/lib.linux-x86_64-3.6/gi
  creating build/lib.linux-x86_64-3.6/gi/repository
  copying gi/repository/__init__.py -> build/lib.linux-x86_64-3.6/gi/repository
  creating build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/GIMarshallingTests.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/keysyms.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/Gdk.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/__init__.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/Gtk.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/Pango.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/Gio.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/GObject.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/GLib.py -> build/lib.linux-x86_64-3.6/gi/overrides
  running build_ext
  pycairo: new API
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 840, in <module>
      main()
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 835, in main
      zip_safe=False,
    File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/setuptools/__init__.py", line 144, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 223, in run
      self.run_command('build')
    File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.6/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 684, in run
      self._setup_extensions()
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 680, in _setup_extensions
      add_pycairo(gi_cairo_ext)
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 665, in add_pycairo
      ext.include_dirs += [get_pycairo_include_dir()]
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 582, in get_pycairo_include_dir
      include_dir = find_path(find_new_api())
    File "/tmp/pip-install-4ex2257z/PyGObject/setup.py", line 532, in find_new_api
      import cairo
  ModuleNotFoundError: No module named 'cairo'
 ----------------------------------------
  >ERROR: Failed building wheel for PyGObject
  Running setup.py clean for PyGObject
  Building wheel for pycairo (setup.py) ... done
  Created wheel for pycairo: filename=pycairo-1.19.1-cp36-cp36m-linux_x86_64.whl size=241262 sha256=ef804c6cbcadb5ae7fe1991338d55ecab1d735cd5edf7dbae4d0ea6eddb2449f
  Stored in directory: /tmp/pip-ephem-wheel-cache-020u07tl/wheels/e5/fa/d2/2ccd226e0c89dd739483929966e3af06cd9f441ac8b5a98997
Successfully built configparser dbus-python psutil pycairo
Failed to build PyGObject
Installing collected packages: configparser, dbus-python, psutil, pycairo, PyGObject, python-periphery
    Running setup.py install for PyGObject ... done
Successfully installed PyGObject-3.28.3 configparser-3.5.0 dbus-python-1.2.8 psutil-5.4.6 pycairo-1.19.1 python-periphery-1.1.1
Enabling and starting systemd service...
All done.<

I've installed all dependencies, including libcairo. After installation lenovo_fix service runs, but i'm not sure, that it works correctly. 

Could you help me with it?

Thank you very much.

KevinNThomas, 2020-03-14:
I also received this error. It looks like the service still may be working but I too wanted to know what was wrong. When I run the install script again I get this output:
```
$ sudo ./lenovo-throttling-fix/install.sh
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Requirement already satisfied: configparser==3.8.1 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (3.8.1)
Requirement already satisfied: dbus-python==1.2.8 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (1.2.8)
Requirement already satisfied: PyGObject==3.32.2 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (3.32.2)
Requirement already satisfied: pycairo>=1.11.1 in ./venv/lib/python3.7/site-packages (from PyGObject==3.32.2->-r requirements.txt (line 3)) (1.19.1)
Enabling and starting systemd service...
All done.
```
Making me think that it may be working fine, but getting some confirmation would be nice.

TheCatster, 2020-08-18:
I'm having this same issue, with the same errors, on Artix runit, difference being your solutions do not work. I've looked through the other closed issues and since this one is still open didn't want to create a new one with the same title. The build fails on `ERROR: Could not build wheels for PyGObject which use PEP 517 and cannot be installed directly
` and I've tried installing everything, modifying the install script, installing with the no pep517 argument and other basic things.

LinuxOnTheDesktop, 2021-12-31:
Same problem - `Could not build wheels for pycairo which use PEP 517 and cannot be installed directly` - on Linux Mint 20.2; and/but the installation fails. Here are some details.

Installing two packages that are, indeed, mentioned in the README as prerequisites seemed to get me a little further. I installed those packages thusly:
```

sudo apt-get install python3-dev
sudo apt install libgirepository1.0-dev
```

Yet, a subsequent run of the install script produced this: 'Package py3cairo was not found in the pkg-config search path.'

Goodness, [Python installations are a pain](https://drewdevault.com/2021/11/16/Python-stop-screwing-distros-over.html).

LinuxOnTheDesktop, 2022-01-01:
I add that I find that version 0.9.2, as against the latest git version, installs and runs without incident.

az0mb13, 2022-01-05:
Solved the PyGObject error with this on arch - `sudo pacman -S gobject-introspection`

LinuxOnTheDesktop, 2022-01-05:
@az0mb13,

thanks. However, I find that, on Mint, gobject-introspection was installed (via apt) already.