erpalma/throttled #185: GPU underclocking does not work on Thinkpad X1 Carbon Gen 6

nszceta, 2020-03-28:
Expected behavior: system crash, lock-up, or freeze upon enabling -999 mV GPU voltage adjustment.

Actual behavior: nothing happens when a -999 mV GPU adjustment is applied. The other parameters seem to work under various stress and mprime test scenarios

System details:
```
Linux x1c6 5.5.11-arch1-1 #1 SMP PREEMPT Sun, 22 Mar 2020 16:33:15 +0000 x86_64 GNU/Linux
```

Command line:
```
$ sudo /usr/lib/throttled/lenovo_fix.py --monitor
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[I] Starting main loop.
[D] Undervolt offsets: CORE: -115.00 mV | GPU: -999.00 mV | CACHE: -115.00 mV | UNCORE: -115.00 mV | ANALOGIO: 0.00 mV
[D] IccMax: CORE: 64.00 A | GPU: 31.00 A | CACHE: 6.00 A
[D] Realtime monitoring of throttling causes:

[AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 518 mV - Package: 1.3 W - Graphics: 0.1 W - DRAM: 0.6 W
```

Configuration file:

```
$ cat /etc/lenovo_fix.conf 
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 1
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: True
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 2
# Disable BDPROCHOT (EXPERIMENTAL)
[... 181 lines cut ...]
[D] core 7 thermal status: thermal limit status = 0
[D] core 7 thermal status: thermal limit log = 0
[D] core 7 thermal status: prochot or forcepr status = 0
[D] core 7 thermal status: prochot or forcepr log = 0
[D] core 7 thermal status: crit temp status = 0
[D] core 7 thermal status: crit temp log = 0
[D] core 7 thermal status: thermal threshold1 status = 0
[D] core 7 thermal status: thermal threshold1 log = 0
[D] core 7 thermal status: thermal threshold2 status = 0
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 0
[D] core 7 thermal status: power limit log = 0
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 1
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 63
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0xf - read 0xf - match OK
[D] CONFIG_TDP_CONTROL - write 0x2 - read 0x2 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd8160 - read 0x42816000dd8160 - match OK
[D] BDPROCHOT - write "0x0" - read "0x0" - match OK
```

erpalma, 2020-04-01:
Can you post a monitor reading under full load?

nszceta, 2020-04-01:
I ran `gputest /benchmark`  twice.

With GPU undervolt set to `-500`:

```
2020-04-01 08:10:26.846: [I] Detected CPU architecture: Intel Kaby Lake (R)
2020-04-01 08:10:26.846: [I] Loading config file.
2020-04-01 08:10:26.857: [I] Starting main loop.
2020-04-01 08:10:26.863: [D] Undervolt offsets: CORE: -115.00 mV | GPU: -500.00 mV | CACHE: -115.00 mV | UNCORE: -115.00 mV | ANALOGIO: 0.00 mV
2020-04-01 08:10:26.864: [D] IccMax: CORE: 64.00 A | GPU: 31.00 A | CACHE: 6.00 A
2020-04-01 08:10:26.864: [D] Realtime monitoring of throttling causes:

2020-04-01 08:10:26.865: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 1126 mV - Package: 18.1 W - Graphics: 0.0 W - DRAM: 0.5 W          
2020-04-01 08:10:27.871: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 1.3 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:28.876: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 1.4 W - Graphics: 0.2 W - DRAM: 0.5 W          
2020-04-01 08:10:29.882: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 524 mV - Package: 0.8 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:30.888: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.8 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:31.892: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 624 mV - Package: 0.8 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:32.898: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.5 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:33.903: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 548 mV - Package: 0.5 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:34.910: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 524 mV - Package: 0.9 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:35.916: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 524 mV - Package: 0.4 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:36.921: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 597 mV - Package: 0.8 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:37.928: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 508 mV - Package: 0.8 W - Graphics: 0.1 W - DRAM: 0.5 W          
2020-04-01 08:10:38.933: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 522 mV - Package: 0.7 W - Graphics: 0.0 W - DRAM: 0.4 W          
2020-04-01 08:10:39.939: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 513 mV - Package: 1.8 W - Graphics: 0.5 W - DRAM: 0.5 W          
2020-04-01 08:10:40.946: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 508 mV - Package: 1.8 W - Graphics: 0.5 W - DRAM: 0.6 W          
2020-04-01 08:10:41.952: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 508 mV - Package: 0.5 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:42.959: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.4 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:43.966: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 509 mV - Package: 1.0 W - Graphics: 0.3 W - DRAM: 0.5 W          
2020-04-01 08:10:44.973: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 509 mV - Package: 0.9 W - Graphics: 0.2 W - DRAM: 0.4 W          
2020-04-01 08:10:45.979: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 513 mV - Package: 0.9 W - Graphics: 0.2 W - DRAM: 0.4 W          
2020-04-01 08:10:46.983: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 509 mV - Package: 1.0 W - Graphics: 0.2 W - DRAM: 0.4 W          
2020-04-01 08:10:47.990: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 509 mV - Package: 0.6 W - Graphics: 0.1 W - DRAM: 0.3 W          
2020-04-01 08:10:48.995: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.5 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:50.002: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.5 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:51.008: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 524 mV - Package: 1.9 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:52.014: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 0.7 W - Graphics: 0.0 W - DRAM: 0.3 W          
2020-04-01 08:10:53.017: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 619 mV - Package: 1.3 W - Graphics: 0.1 W - DRAM: 0.4 W          
2020-04-01 08:10:54.023: [AC] Thermal: OK - Power: OK - Current: OK - Cross-domain (e.g. GPU): OK  ||  VCore: 523 mV - Package: 2.1 W - Graphics: 0.8 W - DRAM: 0.7 W          
[... cut ...]
1116 1112      151  60  0.20    789    105    1.89   0   0    0.00   0   0    0.00   0   0    0.00   0   0 
1091 1083      369   3  1.46   2281   1300   15.59   0   0    0.36   0   0    0.00   0   0    0.00   0   0 
1087 1087       81  65  0.80   1261    837    7.67   0   0    0.53   1   0    0.00   0   0    0.00   0   0 
1088 1088       10  86  0.06    386     93    0.61   0   0    0.00   0   0    0.00   0   0    0.00   0   0 
1082 1082        8  99  0.06    283     88    0.62   0   0    0.00   0   0    0.00   0   0    0.00   0   0 
```

erpalma, 2020-04-02:
Actually the GPU power usage went down by about 2W. There is no official documentation about the actual GPU voltage boundaries, I guess that there might be a maximum ratio between CPU and GPU voltage. We should test this on XTU and see what happens.

nszceta, 2020-04-02:
I take that back. It does work and it instantly crashes with a -1050 mV adjustment. I guess I stopped just short of the crashing threshold when I reported the original bug. I'll post benchmarking results of glmark2 vs voltage offset soon.

erpalma, 2020-04-03:
Hmm, still this sounds me like a bug. It is kind of impossible that your GPU can work at -1V offset voltage. On XTU I get an instant reboot on -500mV

nszceta, 2020-04-03:
[image]

erpalma, 2020-04-05:
So, I've done some tests both on Windows and Linux. I see that you have set throttled like this:

```
CORE: -115
GPU: -999
CACHE: -115
UNCORE: -115
ANALOGIO: 0
```
Now, there is a long standing bug that I need to address (I guess next week) that is related to the labels for these voltage planes. Actually, from Skylake on the UNCORE plane is actually iGPU Unslice and ANALOGIO is SystemAgent, which is mostly the memory controller. Now, iGPU Unslice is a seciton of the GPU which is not dependent on the Execution Units, for instance the video encode/decode units. The processor can selectively enable and disable the execution units, leaving only the Unslice section active, like when you are just playing a movie full screen to save power. I suspect that the iGPU and iGPU Unslice planes are bound and the processor "choose" the higher voltage of the two. If you try lowering both at the same time you should get freezes much much sooner. -900mV offset means that you are actually working near 0V which is clearly impossible. I don't know how you measured the performance in your chart but the vertical axis scale is a bit misleading since there is only a 5% difference from min to max values. Have you repeated the test multiple times? Did you leave time for the laptop to cool down between tests?

nszceta, 2020-04-05:
I kept the laptop hot between tests since there were only a few milliseconds between tests. I documented some more here: https://adamgradzki.com/2020/04/03/thinkpad-gpu-undervolt/

dantempla, 2020-08-07:
I am running the following on my ThinkPad X1C6:

Linux thinkpad-x1c 5.8.0-arch1-1 #1 SMP PREEMPT Mon, 03 Aug 2020 18:43:29 +0000 x86_64 GNU/Linux

```
[UNDERVOLT.BATTERY]
CORE: -140
GPU: -400
CACHE: -140
UNCORE: -100
ANALOGIO: 0

[UNDERVOLT.AC]
CORE: -140
GPU: -400
CACHE: -140
UNCORE: -90
ANALOGIO: 0
```

So far this is running stable. Has there been any new info on the GPU voltage variable?