erpalma/throttled #187: T480 Fixed with N24ET56W (1.31 ) BIOS ?

Hakun1n, 2020-04-08:
Hi, 

I've recently updated BIOS on my T480 and i don't see any throttling-messages in "/var/log/messages" any more (and i was getting a lot of them). Also the throttling messages during system start are gone.

BIOS Information
        Vendor: LENOVO
        Version: N24ET56W (1.31 )
        Release Date: 02/19/2020
        BIOS Revision: 1.31
        Firmware Revision: 1.20

I am on **Fedora 31** and I've used [fwupd](https://linux.com.sg/2019/04/22/fwupd-fedora/) for update.

Maybe someone could confirm this?

vrossum, 2020-04-13:
Maybe you can try to disable throttle fix (see README), reboot, and measure performance.

enzotar, 2020-05-08:
T480 on Ubuntu - not fixed with 1.31