erpalma/throttled #190: Is this fix still relevant for Ubuntu 20.04?

LastStarDust, 2020-04-18:
Hello!
I own a Lenovo Carbon X1 6th gen and I have been using this fix until now for Ubuntu 18.04. I have just switched to Ubuntu 20.04 and I was wondering if the fix was still needed/useful.

I found this [blog](https://mensfeld.pl/2018/05/lenovo-thinkpad-x1-carbon-6th-gen-2018-ubuntu-18-04-tweaks/) online that says that if you get a value of 15 on battery and 3 on AC when running this command
```
sudo rdmsr -f 29:24 -d 0x1a2
```
you do not need the fix anymore, is it true? (I do get those values)

erpalma, 2020-04-18:
There is still no full support for Intel DPTF on Linux so no, 20.04 does not fix this issue. Also undervolt your CPU allows you to reach higher boost frequencies and keep them longer, in particular on very thin laptops like yours. The problem with that MSR is that it can be (and it is) overridden by the EC due to the lack of the support for the adaptive policy on Linux.

LastStarDust, 2020-04-18:
Thank you for the quick and detailed reply. So I will keep on using this fix. Ciao!

closed 2020-04-18 by LastStarDust