erpalma/throttled #191: system crash, X1 carbon 6th, Pop OS 20_04

Kobtul, 2020-04-22:
With these changes on my x1 Carbo 6th gen:

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)! 
[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -50
# Integrated GPU voltage offset (mV)
GPU: -35
# CPU cache voltage offset (mV)
CACHE: -50
# System Agent voltage offset (mV)
UNCORE: -35
# Analog I/O voltage offset (mV)
ANALOGIO: 0

my pc crashes when I attempt to run lenovo_fix.py as root vith correct virtual enviroment with python 3.8. When I tried to install this script, the service fails because of:
ImportError: cannot import name 'DBusGMainLoop' from 'dbus.mainloop

The problem with this crash is, that it persists after reboot and my system crashes again after the boot. It fixes itself after cca 10 minutes. I really do not understand what is going on. I am afraid to run this script again because of that.

erpalma, 2020-04-26:
The issue is just that your CPU does not work with those undervolt values, you need to raise them.

Kobtul, 2020-04-26:
Well why it persisted after reboot? The script was not running as a
service.

erpalma, 2020-04-26:
In theory UEFI should revert those MSR to default, but who knows.. If you power cycle the system you should be fine.

> It fixes itself after cca 10 minutes

What do you mean? 10 minutes idle? Off?

Kobtul, 2020-04-26:
10 minutes of constant boot and crash

erpalma, 2020-04-26:
So, you start throttled with those undervolt values, it crashes, you *turn off* the notebook, after turning it back on it still crashes for about 10 minutes?

Kobtul, 2020-04-26:
Yes, that is why I am afraid to run this script again. I am on UEFI boot.

erpalma, 2020-04-26:
Ok if you power cycled your notebook then for sure UEFI is restoring power planes to default. MSR are volatile, are you 100% sure that throttled is not started automatically at boot by systemd?

Kobtul, 2020-04-26:
Well I created manually the python environment and ran this script. When I
tried to install it as a service I got the described error. Also it crashed
always so quickly that it was impossible for me to switch to the terminal.
If I understand correctly, if it was running at boot it would not fixed
itself right? The whole behavior does not make any sense to me.

nariox, 2020-04-30:
You should have a tiny button in the back to reset the EC. Press it with something thin and hold it for 5 seconds, that should reset everything back to "normal".

erpalma, 2020-04-30:
That button is also resetting UEFI to defaults! Still, if you have troubles with undervolt just set all values to zero ;)

Kobtul, 2021-02-22:
Ok, after half a year I gave it another go.
The problem with the install script was python 3.8, so I installed python 3.6 and changed /usr/bin/python3 -m venv venv
to /usr/bin/python3.6 -m venv venv

I started again from 0 undervolt, now I am at -80 on CPU and -10 on GPU. Everything is fine. I probably got too scared when my laptop rebooted multiple times.