erpalma/throttled #192: works on Lenovo T460 (i5-6300U)

eleon216, 2020-04-22:
(if this is the wrong way to report this, please feel free to close/delete this issue)

I used the tool on my T460 (i5-6300U with iGPU, Pop!_OS 19.10 (Ubuntu-derivat)) to increase the cTDP from 15W to 25W

The increased TDP is (only) useful  if CPU and GPU are under load at the same time. 

cTDP: 15W
CPU-Stresstest: Both CPU-Cores stay at 2.9GHz  Temps stay below 70°C (14.5-14.9W)
CPU+GPU-Stresstest: CPU-frequencies drop to 2.2GHz to stay below 15W

cTDP: 25W
CPU-Stresstest: again 2.9GHz around 15W
CPU+GPU-Stresstest: CPU-frequencies at 2.8GHz (litte throttle to stay below 25W)
But 25W are too much for the coolingsystem. I set the max-temp to 80°C, and after a few minutes it's throttling a little to stay below 80° (2.7-2.75GHz 23-24W) 
The T460 has only one, quite long heatpipe (the T460s which ships with a cTDP of 25W, has a shorter heatpipe,  some models even with a dual heatpipe-arragement)  

Maybe undervolting will help, but +500MHz under CPU- and GPU-load is not bad ;)

erpalma, 2020-04-26:
Actually there is an additional setting that I need to implement that tell the CPU to prioritize CPU or iGPU clocks under load. Also 80° are super safe, I guess you can push it a bit further.

eleon216, 2020-04-27:
Thanks for the reply. 
No need to keep this issue open, I'm quite happy with the results. 
I opened it more for documentation, because the Thinkpad T460 (or any other Thinkpad with 6th-gen Intel-CPUs) was listed as working/tested.  
The increased TDP  works well and is a great improvement over the stock TDP, and apart from running actual stresstests (CPU and GPU simultaniously) the coolingsystem can deal with generated heat just fine. 

But this additional setting sounds interesting, and I'm would be happy to test/use it. But probably no real need for it in my case, as my problems are already fixed with the increased TDP.

closed 2020-04-27 by eleon216