erpalma/throttled #194: Not working with Python3.8 in Ubuntu 20.04

pdelteil, 2020-04-26:
I updated to Ubuntu 20.04 and the fix stopped working. Tried to install it again but I discovered that Ubuntu 20.04 packs python3.8, I was getting this errors:

```
> sudo ./install.sh 
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/opt/lenovo_fix/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

```

You will need to install python3.7 to make it work (don't uninstall python3.8 it might break your system) 

Steps:

```
sudo add-apt-repository ppa:deadsnakes/ppa 
sudo apt-get install python3.7 python3.7-venv

cd /opt/lenovo_fix/venv/bin
rm python3  #removes the link to python3.8
sudo ln -s  /usr/bin/python3.7 python3 

```

Then the installation will work.

erpalma, 2020-04-26:
The problem seems not related to python3.8 but rather to `python3-venv` not installed.

pdelteil, 2020-04-26:
That's not the case because python3-venv was installed.

Did you test it?

erpalma, 2020-04-27:
Just tested on Ubuntu 20.04 and it works just fine. `python3-venv` is clearly not installed on your system.

closed 2020-04-27 by erpalma

marto87, 2020-04-27:
Same problem here.
```
$ sudo ./install.sh 
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/opt/lenovo_fix/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']
```

And python3-venv is installed as you can see:
```
$ dpkg -l | grep python3-venv
ii  python3-venv                               3.8.2-0ubuntu2                                   amd64        pyvenv-3 binary for python3 (default python3 version)
```
I tried this solution that fixes lenovo_fix but breaks almost anything else using python
```
rm python3  #removes the link to python3.8
sudo ln -s  /usr/bin/python3.7 python3 
```

I'm sure there's something else that we can do, but I think the problem isn't python3-venv not installed.

pdelteil, 2020-04-27:
> Just tested on Ubuntu 20.04 and it works just fine. `python3-venv` is clearly not installed on your system.

I'm offended by the fact you think I wasn't able to check if python3-venv was installed or not, that was the simplest solution but didn't solve the problem. 

@marto87  has the same problem.  Maybe you could fix it following the instructions I just posted?

marto87, 2020-04-27:
>  Maybe you could fix it following the instructions I just posted?
Yes, it fixed the lenovo_fix service but it breaks anything else related to python in my system.

xconverge, 2020-04-27:
"After installing the python3-venv
package, recreate your virtual environment."

I fixed this by removing /opt/lenovo_fix/venv/ and installing again, thus recreating the virtual environment

marto87, 2020-04-27:
Thank you @xconverge I did that and now it seems its working again, but it throws some errors during installation:

```
Copying config file...                                                                                                                                                                                             
Config file already exists, skipping.                                                                                                                                                                              
Copying systemd service file...               
Building virtualenv...               
Collecting configparser==3.8.1
  Using cached configparser-3.8.1-py2.py3-none-any.whl (22 kB)                                                                                                
Collecting dbus-python==1.2.8                                                                                          
  Using cached dbus-python-1.2.8.tar.gz (787 kB)                                                                       
Collecting PyGObject==3.32.2                    
  Using cached PyGObject-3.32.2.tar.gz (698 kB)                                                                                                                                                                                               
  Installing build dependencies ... done                                                                                                                                                                                                      
  Getting requirements to build wheel ... done                                                                                                                                                                     
    Preparing wheel metadata ... done            
Collecting pycairo>=1.11.1                                                                                             
  Using cached pycairo-1.19.1.tar.gz (205 kB)                                                                                                                                                                      
Building wheels for collected packages: dbus-python, PyGObject, pycairo                                                                                                                                            
  Building wheel for dbus-python (setup.py) ... error                                                                                                         
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-zd79h_ju/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-zd79h_ju/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'
"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-y8kysq4e                                                    
       cwd: /tmp/pip-install-zd79h_ju/dbus-python/
  Complete output (6 lines):            
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]                                                                                       
     or: setup.py --help [cmd1 cmd2 ...]                                                                                                                                                                                                      
     or: setup.py --help-commands                                                                                                                                                                                                             
     or: setup.py cmd --help                     
                                                
  error: invalid command 'bdist_wheel'                                                                                                                                                                                                        
  ----------------------------------------                                                                                                                                                                                                    
  ERROR: Failed building wheel for dbus-python                                                           
  Running setup.py clean for dbus-python                                                                 
  Building wheel for PyGObject (PEP 517) ... done                                                                      
  Created wheel for PyGObject: filename=PyGObject-3.32.2-cp38-cp38-linux_x86_64.whl size=872954 sha256=f5c0424afcd24ea0b5f4fcd12bd7d67585db230dffc21d2c319e51ee6b178d28                                                                                                                                                      
  Stored in directory: /root/.cache/pip/wheels/e7/26/aa/27ee2ac4074e4d395cf32e9b4a9a1997f6dd68d352c14d62e0                                                                                                                                                                                                                   
  Building wheel for pycairo (setup.py) ... error
[... cut ...]
Installing collected packages: configparser, dbus-python, pycairo, PyGObject                                           
    Running setup.py install for dbus-python ... done                                                                                                         
    Running setup.py install for pycairo ... done                              
Successfully installed PyGObject-3.32.2 configparser-3.8.1 dbus-python-1.2.8 pycairo-1.19.1                                                                   
Enabling and starting systemd service...                                       
All done. 
```

enzotar, 2020-05-21:
@marto87 Same error message here

enzotar, 2020-05-21:
@marto87 switched to Python 3.6 and it worked
https://hackersandslackers.com/multiple-versions-python-ubuntu/