erpalma/throttled #196: The mandatory "Update_Rate_s" parameter is missing.

Lusitaniae, 2020-05-01:
```
~/workspace/lenovo-throttling-fix$ sudo ./lenovo_fix.py --debug
[I] Detected CPU architecture: Intel Haswell
[I] Loading config file.
[E] The mandatory "Update_Rate_s" parameter is missing.

```

This is after following the Ubuntu installatnowion instructions.

```
sudo systemctl status lenovo_fix
● lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2020-05-02 03:13:33 +07; 6s ago
  Process: 28449 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
 Main PID: 28449 (code=exited, status=1/FAILURE)

May 02 03:13:32 q systemd[1]: Started Stop Intel throttling.
May 02 03:13:33 q python3[28449]: [I] Detected CPU architecture: Intel Haswell
May 02 03:13:33 q python3[28449]: [I] Loading config file.
May 02 03:13:33 q python3[28449]: [E] The mandatory "Update_Rate_s" parameter is missing.
May 02 03:13:33 q systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
May 02 03:13:33 q systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.

```

Lusitaniae, 2020-05-02:
Think I created the conf file in /etc with the undervolt block only from the Readme. 

Installing the original one from the repo solved it.

closed 2020-05-02 by Lusitaniae

Tgodmuna, 2025-01-10:
Thanks for the suggestion! I had a similar issue, and installing the original config file from the repo solved it. After replacing my custom config with the one from the repo, the service started working correctly.

For anyone else facing the same issue, make sure you're using the original config file instead of a custom one to avoid the 'missing parameters' error.