erpalma/throttled #197: How to setup on Ubuntu 20.04

enzotar, 2020-05-08:
I upgraded from Ubuntu 18.04 to 20.04, and most of the tools seems to have been deprecated or not working. It was working well previously. Ubuntu 18.04 on T480.

sudo modprobe msrsudo
modprobe: FATAL: Module msrsudo not found in directory /lib/modules/5.4.0-29-generic

s-tui is giving python related warning errors

sudo ./install.sh is asking for a package even though it is installed
_"The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package...Failing command: ['/opt/lenovo_fix/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']"_ 

Any ideas to make it work on the Ubuntu 20.04?

enzotar, 2020-05-08:
got
_sudo rdmsr -f 29:24 -d 0x1a2_
running after 
sudo modprobe msr instead of msrsudo

enzotar, 2020-05-08:
Followed these post below instead and it works, but I have to restart thermald after a reboot.
https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4028489?page=34#4637873

1. Check TLP options

 

In /etc/tlp.conf (TLP >= 1.3) or /etc/default/tlp (TLP < 1.3), be sure to have

CPU_ENERGY_PERF_POLICY_ON_AC=balance_performance
CPU_ENERGY_PERF_POLICY_ON_BAT=balance_power
You can change to

CPU_ENERGY_PERF_POLICY_ON_BAT=balance_performance
if you want the perfs to be maximal on BAT too. Don't use the (1.3) power option default on BAT, it's makes the cpu stay at its lower frequency levels, except if you want that of course...

 

2. Create a thermald configuration file

 

In :

/etc/thermald

you will include some thermald configuration files (thermal rules) by running :

sudo apt install dptfxtract
The one that will be used by thermald after restart is *.auto .

 

3. Change the systemd thermald service

 

Modify :

/lib/systemd/system/thermald.service

to add the --workaround-enabled option to thermald's start :

[Unit]
Description=Thermal Daemon Service

[Service]
Type=dbus
SuccessExitStatus=1
BusName=org.freedesktop.thermald
ExecStart=/usr/sbin/thermald --no-daemon --dbus-enable --workaround-enabled

[Install]
WantedBy=multi-user.target
Alias=dbus-org.freedesktop.thermald.service
4. Restart

 

Restart the systemd thermald service :

sudo systemctl restart thermald
or reboot your computer.

 

5. Check that the workaround does remove excessive throttling

 

Start stressing with the stress option after launching :

s-tui
For me (T480s, i7 8550u) after some stress time :

balance_performance : cpu freq 2900 MHz, cpu power 25 W, fan 5080 rpm
balance_power : cpu freq 2300 MHz, cpu power 14 W, fan 0 rpm

erpalma, 2020-05-10:
`msrsudo` module does not exists. It is just `msr` in any kernel version. Also you have to disable thermald as stated in the readme.

cschroeder-351, 2020-05-13:
To answer part of the original Question

> sudo ./install.sh is asking for a package even though it is installed
The virtual environment was not created successfully because ensurepip is not
available. On Debian/Ubuntu systems, you need to install the python3-venv
package...Failing command: ['/opt/lenovo_fix/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

There seems to be a problem with Ubuntu 20.04 and virtualenv, i solved it by moving the install directory to my home directory and edited the Service file accordingly, also see:
[https://stackoverflow.com/questions/61596003/pip3-is-unable-to-create-virtual-environment-on-ubuntu-20-04-lts-on-windows-10-b](url)

enzotar, 2020-05-21:
Got it running by switching to Python 3.6

LinuxOnTheDesktop, 2020-07-10:
@erpalma 

Any chance of a fix within your program, or at least of some instructions that are easier to follow? I am unable to follow the advice people have posted above - because I do not understand it well enough.

erpalma, 2020-07-10:
I've just tested (again) the install script on a fresh Ubuntu 20.04 and it just works. I've added `wheel` to the install script to avoid the failed bdist_wheel message.

LinuxOnTheDesktop, 2020-07-10:
Thanks (and nice palm tree) but on my newly-upgraded Mint 20 the following happens.

```
$ ./install.sh
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
cp: cannot create regular file '/etc/systemd/system/lenovo_fix.service': Permission denied
nicholas@x1 ~/git/lenovo-throttling-fix/lenovo-throttling-fix $ sudo ./install.sh
[sudo] password for nicholas: 
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/opt/lenovo_fix/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

 apt list -a python3-venv
Listing... Done
python3-venv/focal,now 3.8.2-0ubuntu2 amd64 [installed]

python3-venv/focal 3.8.2-0ubuntu2 i386

```
This may not be your fault but I'd appreciate any help.

erpalma, 2020-07-10:
Do you have multiple Python3 versions installed? I guess you have something wrong with python on your distro.

LinuxOnTheDesktop, 2020-07-10:
Thanks. I see this:

```
$ python3
Python 3.8.2 (default, Apr 27 2020, 15:53:34)
[GCC 9.3.0] on linux
```

Perhaps VirtualBox - the cause of the bulk of my computer problems, and across platforms - is to blame. I think it installed a 'python-is-python2' package. Also, I think that Mint is planning to give users - but has not done so yet - a 'python-is-python3' package. However, I think your install script calls python_3_ explicitly . .

erpalma, 2020-07-10:
Yes it does, but there is clearly something wrong with your python setup that mess things up.

LinuxOnTheDesktop, 2020-07-10:
Thanks. I hope you will be happy to leave this issue open, even if the problem is not with your software. That way, I and others affected have a slightly better chance of finding a solution.

LinuxOnTheDesktop, 2020-07-11:
[Relevant bug report](https://github.com/rstudio/reticulate/pull/370) -that suggests there is something to be done here as against upstream?

erpalma, 2020-07-11:
> [Relevant bug report](https://github.com/rstudio/reticulate/pull/370) -that suggests there is something to be done here as against upstream?

Can you elaborate on how this is related?

LinuxOnTheDesktop, 2020-07-11:
I can, but only a little. That report is about a problem that a program had in detecting a python virtual environment; and the report gives a solution too. That account of that problem suggests somewhat that it can be by dint of something that a program itself does, that the environment is not recognised (or at least that a program can work around an upstream problem). I thought that that might be the case here with Throttled. And I thought perhaps the solution given in the report would apply. However, I have no real understanding of any of this.

LinuxOnTheDesktop, 2020-07-11:
FIXED - as follows.

Remove the program's directory in opt - a directory that contained 'virtualenv' stuff.
Replace my copy of the programs install script with the latest one.
Run the install script.

Kobtul, 2021-02-22:
I also had to switch from python 3.8 to python 3.6 for the service to get it working. I was getting this error with python 3.8:

ImportError: cannot import name 'DBusGMainLoop' from 'dbus.mainloop.glib' (/opt/lenovo_fix/venv/lib/python3.8/site-packages/dbus/mainloop/glib.py