erpalma/throttled #198: Dell XPS 9365 i7-8550u Current Limited

ToRvaLDz, 2020-05-16:
Hi,
please help me to understand if everything is ok.

I have a DELL XPS 13 9565 Intel i7 8550u, under normal operations no stress AC powered, I got:

[image]

Current is always limited and the cpu never reach the theoretical 4.0ghz.

While stressing the cpu the situation change:

[image]

My config:
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 25
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 25
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 97
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 25
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 25
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 97
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: true
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: true

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)! 
[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -90
#-105
# Integrated GPU voltage offset (mV)
GPU: -55
#-55
# CPU cache voltage offset (mV)
CACHE: -90
#-90
# System Agent voltage offset (mV)
UNCORE: -85
#-85
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)!
[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -90
#-105
# Integrated GPU voltage offset (mV)
GPU: -55 
#-55
#-85
# CPU cache voltage offset (mV)
CACHE: -90
#-90
#-105
# System Agent voltage offset (mV)
UNCORE: -85
#-85
# Analog I/O voltage offset (mV)
ANALOGIO: 0

#[ICCMAX.AC]
# # CPU core max current (A)
#CORE: 24
# # Integrated GPU max current (A)
#GPU: 24
# # CPU cache max current (A)
#CACHE: 4

# [ICCMAX.BATTERY]
# # CPU core max current (A)
#CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 
```

My governor is set to performance.

What's wrong?

Thank you

chaoyi, 2020-05-16:
Hi, I am just a happy user of throttled.

It seems to me BIOS is limiting the current.  I would try Cinebench on Windows and compare the score with those on notebookcheck.net.

ToRvaLDz, 2020-05-18:
> Hi, I am just a happy user of throttled.
> 
> It seems to me BIOS is limiting the current. I would try Cinebench on Windows and compare the score with those on notebookcheck.net.

Hi,
thank you for your suggestion.

erpalma, 2020-05-21:
You can try to increase ICCMAX but you might damage the voltage regulators if they are not designed to handle high current.