erpalma/throttled #199: Install failed because of "wheel error"

hellofaduck, 2020-05-20:
Installing on fresh Pop OS 20.04 on t480
Followed readme step by step and after executing install i have this:
```
  Building wheel for dbus-python (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2h2_e_y7/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2h2_e_y7/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-kwpy38o8
       cwd: /tmp/pip-install-2h2_e_y7/dbus-python/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for dbus-python
  Running setup.py clean for dbus-python
  Building wheel for pycairo (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2h2_e_y7/pycairo/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2h2_e_y7/pycairo/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-vg86vj70
       cwd: /tmp/pip-install-2h2_e_y7/pycairo/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for pycairo
```
I try to install wheel command by pip3 install wheel, it said that already installed and do nothing... tried different workarounds found on google but it wont work ((

erpalma, 2020-05-21:
This is just telling you that the installer is not able to build the wheel for pycairo. The venv should be fine though.

hellofaduck, 2020-05-23:
Moved to manjaro, problem gone

closed 2020-05-23 by hellofaduck

Kobtul, 2021-02-22:
Installing python3.6-dev solved this for me on Pop OS.