erpalma/throttled #201: It seems like it's actually possible to use Throttled with Secure Boot enabled

zeroxfourc, 2020-05-26:
Just thought I'd let you know that I've managed to run Throttled with Secure Boot enabled. I apologise if this isn't the right place to post this, but here's how I did it:

The file `/sys/kernel/lockdown/lsm` contains information about what LSM:s are in use. On my system it contained `capability,lockdown,yama,apparmor`, though I'd imagine it varies depending on your setup. Take the list you get from your file, remove `lockdown` and add it as a kernel parameter, like this: `lsm=capability,yama,apparmor`. Reboot, enable Secure Boot and Throttled should work!

It would be nice if people could test this so we know it doesn't only work on my machine.

erpalma, 2020-05-26:
Well you are actually right. The problem is not secure boot by itself but the lockdown security module. I guess we could add this on the readme. Would you please make a PR to explain this?

zeroxfourc, 2020-05-27:
Sure!

closed 2020-06-22 by lolomokakus

dgutov, 2020-07-16:
Doesn't seem to work here. I changed `/etc/default/grub`, ran `sudo update-grub`, rebooted.

Now the kernel parameter is seemingly okay:

```
$ cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-5.4.0-40-generic root=/dev/mapper/ubuntu--vg-root ro quiet splash lsm=capability,yama,apparmor vt.handoff=7
```

And yet `/sys/kernel/lockdown/lsm` still says `lockdown,capability,yama,apparmor`, and the MSR access doesn't work.

jayarmstrong, 2021-04-06:
> And yet `/sys/kernel/lockdown/lsm` still says `lockdown,capability,yama,apparmor`, and the MSR access doesn't work.
Yep. My (opensuse) system also ignores the lsm= parameter and enables lockdown anyway. There seems to be no way to enable Secure Boot but disable Lockdown.

Perhaps the documentation needs revision for this?

AgenttiX, 2022-10-25:
It seems that I can't disable the kernel lockdown at all on ThinkPad T480 with Ubuntu 22.04.1.
```
$ sudo nano /etc/default/grub
# Added: GRUB_CMDLINE_LINUX_DEFAULT="quiet splash lsm=capability,landlock,yama,apparmor"
$ sudo update-grub
# Reboot
$ cat /sys/kernel/security/lsm
lockdown,capability,landlock,yama,apparmor%
```

Disabling shim validation did not help:
```
$ sudo mokutil --disable-validation
# Reboot, disabled Secure Boot validation in the shim menu
$ sudo mokutil --sb-state
SecureBoot enabled
SecureBoot validation is disabled in shim
$ cat /sys/kernel/security/lsm
lockdown,capability,landlock,yama,apparmor%
```

Disabling Secure Boot from the UEFI/BIOS settings and restricting the list of LSM modules further to `lsm=capability,yama` as instructed in the throttled readme did not help either:
```
$ sudo mokutil --sb-state
SecureBoot disabled
$ cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-5.15.0-52-generic root=/dev/mapper/crypto--vg-crypto--root ro quiet splash lsm=capability,yama vt.handoff=7
$ cat /sys/kernel/security/lsm
lockdown,capability,yama%
```

How can I disable the kernel lockdown to get throttled working?

AgenttiX, 2022-10-25:
Sorry, my bad. It seems that `/sys/kernel/security/lsm` has `lockdown` in it even when lockdown is disabled, and instead `/sys/kernel/security/lockdown` tells whether it's actually enabled. Either disabling shim validation with `sudo mokutil --disable-validation` or disabling Secure Boot in the UEFI/BIOS settings seems to be sufficient, resulting in disabled lockdown:
```
$ cat /sys/kernel/security/lockdown
[none] integrity confidentiality
```

However, adjusting the lsm kernel parameter has no effect on the lockdown and is therefore not sufficient to disable it.