erpalma/throttled #203: Failed to build dbus-python pycairo

kriggus, 2020-06-17:
Hi, thanks for a great tool! Worked great on my thinkpad l390 ubuntu 18.04. But I get this error trying to reinstall after update to 20.04.

```
$ sudo ./install.sh 
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Collecting configparser==3.8.1
  Using cached configparser-3.8.1-py2.py3-none-any.whl (22 kB)
Collecting dbus-python==1.2.8
  Using cached dbus-python-1.2.8.tar.gz (787 kB)
Processing /root/.cache/pip/wheels/e7/26/aa/27ee2ac4074e4d395cf32e9b4a9a1997f6dd68d352c14d62e0/PyGObject-3.32.2-cp38-cp38-linux_x86_64.whl
Collecting pycairo>=1.11.1
  Using cached pycairo-1.19.1.tar.gz (205 kB)
Building wheels for collected packages: dbus-python, pycairo
  Building wheel for dbus-python (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-m0ya1zgh/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-m0ya1zgh/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-lrd1qqfd
       cwd: /tmp/pip-install-m0ya1zgh/dbus-python/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for dbus-python
  Running setup.py clean for dbus-python
  Building wheel for pycairo (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-m0ya1zgh/pycairo/setup.py'"'"'; __file__='"'"'/tmp/pip-install-m0ya1zgh/pycairo/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-0z7c6dzo
       cwd: /tmp/pip-install-m0ya1zgh/pycairo/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for pycairo
  Running setup.py clean for pycairo
Failed to build dbus-python pycairo
Installing collected packages: configparser, dbus-python, pycairo, PyGObject
    Running setup.py install for dbus-python ... done
    Running setup.py install for pycairo ... done
Successfully installed PyGObject-3.32.2 configparser-3.8.1 dbus-python-1.2.8 pycairo-1.19.1
Enabling and starting systemd service...
All done.
```

Any ideas?

erpalma, 2020-06-18:
See #199.

closed 2020-06-18 by erpalma

kriggus, 2020-06-26:
I just ignored the error msg and it seems to work.

And again, thanks for the great tool! It makes my l390 usable on a 4k display in ubuntu! :-----)