erpalma/throttled #205: L380 Overheating

HSPDev, 2020-07-02:
The tool works fine and my L380 is much more usable than ever.

However I've experienced 3 thermal shutdowns in a week, but it is always when I'm doing something "stupid" as watching 2.7k Youtube through DisplayLink (uses CPU rendering) or stress testing.

I've tried undervolting to reduce temperature, but it still trips out. 

Any suggestions on how to debug/resolve this?

I'm gonna do the basics and clean the laptop + apply new thermal grease.

This is my config:

```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 42
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 80
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 42
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 42
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 80
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)! 
[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -20
# Integrated GPU voltage offset (mV)
GPU: -20
# CPU cache voltage offset (mV)
CACHE: -20
# System Agent voltage offset (mV)
UNCORE: -20
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)!
[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -20
# Integrated GPU voltage offset (mV)
GPU: -20
# CPU cache voltage offset (mV)
CACHE: -20
# System Agent voltage offset (mV)
UNCORE: -20
# Analog I/O voltage offset (mV)
ANALOGIO: 0

# [ICCMAX.AC]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

# [ICCMAX.BATTERY]
# # CPU core max current (A)
# CORE: 
# # Integrated GPU max current (A)
# GPU: 
# # CPU cache max current (A)
# CACHE: 

```

tamalban, 2020-11-10:
Hi, I also have an L380. I am able to run it smoothly with some changes.

The default power limits in the .conf file are too high for L380. That has also causes thermal shutdowns for me, when the CPU gets  close to 95C for a couple of minutes. The heatsink of the L380 is too weak to dissipate heat.

I use power limits of 25W/15W and trip temperature of 85C, along with disabling BD_PROCHOT and that solves all issues for me.
I am also undervolting Cpu and Cache with -80mV and iGPU with -50mV. 

Try these changes and see if it works for you. Remember to fully uninstall the tool (including deleting the .conf file in /etc), and then install as usual...

HSPDev, 2020-11-26:
Good thing you reminded me to check this thread @tamalban 

Honestly, I removed this tool entirely and installed all the available BIOS updates.
Lenovo has mostly fixed the issue and I'm not throttling anymore if I use the L380.

I ended up replacing the unit for a P53 with i7 for even more power though, but that's just because I needed an upgrade.
Much better thermal design in general in the P series, but holy **** it sounds like a jet engine. :open_mouth:

closed 2020-11-26 by HSPDev

f100024, 2020-12-17:
Hi guys @HSPDev @tamalban ! I'm using L380 on ubuntu. I've some issues looks like similar to yours, could you please help me with it:

- How did you detect that shutdown was related to thermal issue? For my case laptop just hangs with black screen and other artifacts on screen. My old lenovo e320 when got high temperature, were rebooted, but L380 hangs. Did you have similar symptoms?

- You've said that never used win and updated bios, if I clearly understood, you have used boot image provided by lenovo. If yes, have you faced with any issues or found some corner cases. I'm asking this because didnt' found enough information and manual for updating from boot image by linux users.

tamalban, 2020-12-19:
@f100024 Hi Artem,
-I checked the logs and there were complaints just before thermal shutdown. Also I constantly monitor CPU temperature using a widget and notice these problems only when the notebook overheats. Have you tried using any other distro? I personally use Kubuntu, and had other issues with the Gnome version (regular ubuntu). Kubuntu runs smooth for me.
- I dual boot Win 10 and Kubuntu, and use the latest BIOS. I have re-installed windows perhaps 4-5 times already, so I am not on Lenovo's original image. I recommend keeping Windows 10 as dual boot on a small partition, just for firmware updates. I also found that battery charge thresholds work the best when set in Windows. Setting the same in TLP caused conflicts for me.
- 
- @HSPDev Good to hear that it is working well for you. Even with the latest updates, if I don't disable BDPROCHOT, everything is fine until I run anything that stresses the iGPU. Even running CS1.6 in Windows locks down the clockspeed. So I continue using these tools.

daviewales, 2024-07-08:
I was able to update the BIOS using fwupd on Linux Mint 21.3, following these instructions:
https://www.cyberciti.biz/faq/thinkpad-update-firmware-on-linux-x1-extreme-p1-gen2/

I was on a 2018 version, and now I'm on a 2024 BIOS version. (1.37)

Pre-update, my Geekbench 6 score on battery power was 1139 single-core, and 2941 multi-core.
Post-update, I got 1167 single-core and 3678 multi-core on battery.
So it seems the BIOS update improved multi-core performance.

However, on power, I get only 616-630 single-core and 2773-2846 multi-core. (post-BIOS upgrade. Didn't think to test on power pre-upgrade.)

I'll check if throttled improves anything further.