erpalma/throttled #216: Input/output error on readmsr

YiannisGkoufas, 2020-09-29:
Hi there! I have a lenovo T480 and the process fails with the following logs:

```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
Traceback (most recent call last):
  File "/opt/lenovo_fix/lenovo_fix.py", line 882, in <module>
    main()
  File "/opt/lenovo_fix/lenovo_fix.py", line 824, in main
    set_hwp(config.getboolean('AC', 'HWP_Mode', fallback=False))
  File "/opt/lenovo_fix/lenovo_fix.py", line 523, in set_hwp
    cur_val = readmsr(0x774, cpu=0)
  File "/opt/lenovo_fix/lenovo_fix.py", line 175, in readmsr
    raise e
  File "/opt/lenovo_fix/lenovo_fix.py", line 165, in readmsr
    val = struct.unpack('Q', os.read(f, 8))[0]
OSError: [Errno 5] Input/output error
```

erpalma, 2020-09-29:
Do you have secure boot enabled?

YiannisGkoufas, 2020-09-29:
Hi sorry, not sure what you mean :( ( total n00b here)
That is all the logs I get using the --debug flag
And I had secure boot disabled. When I enabled it I got:
```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 21
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[D] cpu platform info: feature ppin cap = 0
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 2
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[E] Unable to write to MSR. Try to disable Secure Boot and check if your kernel does not restrict access to MSR.
```
which makes sense :D

erpalma, 2020-09-29:
Can you try running `rdmsr 0x774` as root?

YiannisGkoufas, 2020-09-29:
`rdmsr: CPU 0 cannot read MSR 0x00000774`

YiannisGkoufas, 2020-10-01:
By the way @erpalma THANK YOU!
Disabled only this line, run the file and helped me get my CPU back to working state!!!!
Before your script, it was throttling to 200MHz (!!!) and after some time increasing(...) to 400MHz.
Your utility helped me to bring the CPU back to normal.

kadru, 2020-11-25:
Hi,

I have the same problem (t480 and secure boot disable), but I commented the line 824 and it started  working. I don't sure if correct what I've done.

goodboy, 2021-03-27:
I have this same problem on kernel `5.11.9`

```
 >>> sudo rdmsr 0x774
rdmsr: CPU 0 cannot read MSR 0x00000774
```

I can also confirm that @kadru's hack worked.

Commenting [the `set_hwp()` line](https://github.com/erpalma/throttled/blob/master/lenovo_fix.py#L933) solves the issue.

No clue if this is correct but it does seem to allow the script to run as previous.

@erpalma, I'm happy to make a PR if there is some adjustment I can make / test to make this more correct.

erpalma, 2021-03-27:
@goodboy this issue should already be fixed in 3f3b4420