erpalma/throttled #217: Doesn't compile with Python 3.9 (FC33)

tokyovigilante, 2020-10-01:
Hi, Fedora 33 is shipping with Python 3.9, which has removed support for ` DBusPyFloatBase_Type.tp_print`

```
  libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/tmp/pip-install-srsl1fye/dbus-python -include config.h -I/tmp/pip-install-srsl1fye/dbus-python/include -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/python3.9 -fno-strict-aliasing -Wdeprecated-declarations -Wno-duplicated-branches -Wno-redundant-decls -Wno-switch-default -Wno-write-strings -Wall -Wextra -Wundef -Wnested-externs -Wpointer-arith -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wno-unused-parameter -Wno-missing-field-initializers -Wdeclaration-after-statement -Wformat=2 -Wold-style-definition -Wcast-align -Wformat-nonliteral -Wformat-security -Wsign-compare -Wstrict-aliasing -Wshadow -Winline -Wpacked -Wmissing-format-attribute -Wmissing-noreturn -Winit-self -Wmissing-include-dirs -Wunused-but-set-variable -Warray-bounds -Wimplicit-function-declaration -Wreturn-type -Wswitch-enum -Wno-error=duplicated-branches -Wno-error=redundant-decls -Wno-error=switch-default -Wno-error=write-strings -Wno-error=unused-parameter -Wno-error=missing-field-initializers -g -O2 -MT dbus_bindings/abstract.lo -MD -MP -MF dbus_bindings/.deps/abstract.Tpo -c /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c  -fPIC -DPIC -o dbus_bindings/.libs/abstract.o
  /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c: In function ‘dbus_py_init_abstract’:
  /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c:793:25: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
    793 |     DBusPyBytesBase_Type.tp_print = NULL;
        |                         ^
  /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c:804:25: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
    804 |     DBusPyFloatBase_Type.tp_print = NULL;
        |                         ^
  /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c:808:24: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
    808 |     DBusPyLongBase_Type.tp_print = NULL;
        |                        ^
  /tmp/pip-install-srsl1fye/dbus-python/dbus_bindings/abstract.c:812:23: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
    812 |     DBusPyStrBase_Type.tp_print = NULL;
        |                       ^
  make[2]: *** [Makefile:1391: dbus_bindings/abstract.lo] Error 1
```

closed 2020-10-06 by erpalma

tokyovigilante, 2020-10-06:
Working great now, thanks!

LinuxOnTheDesktop, 2020-10-12:
Er, will this change stop the program from running on machines that have a version of Python earlier than 3.9?

erpalma, 2020-10-12:
Nope, I've updated the dbus module for supporting (also) Py3.9

timrichardson, 2020-11-01:
Oh, master for me didn't work, what does "I've updated the dbud module for supporting (also) Py3.9" mean ... on Pypi?

commit d6aeb685 (2020-10-06): update dependencies, fix #217