erpalma/throttled #219: Does the issue this solves still exist? And on newer devices?

jgarvin, 2020-10-08:
I originally used this tool for my X1 Gen 6. Doing googling I'm getting some conflicting reports, where Lenovo allegedly promised a firmware update to fix but I can't find a version number.

* Did it ever happen?
* If it did, does it obviate this tool?
* Does this problem still exist on X1 Gen 8?

Thanks

depate, 2020-10-19:
I read as well that Lenovo wanted to fix something. See [0] of this [mail](https://lore.kernel.org/lkml/20200714121955.GA2080@chrisdown.name/)

However this [Thread](https://lore.kernel.org/lkml/e33ebe71317b0a09ca40e97fb76ff57313c0444c.camel@linux.intel.com/) also concludes that changes from [Matthew](https://github.com/mjg59) should've been merged to `thermald` in Kernel 5.8. 

I did not try to find them, yet, I'll try later today. But maybe @erpalma is in the know. I really like `throttled`'s lenovo_fix and it can do much more than I need, but if standard `thermald` suffices, I'd been eager to switch.

LinuxOnTheDesktop, 2020-11-04:
I have an X1CG6, with kernel 5.9.0, and so far as I know - and I do skim the kernel change logs - we do still need the good Mr Palma's fix.

a-vrma, 2020-12-09:
have tested with s-tui on Kernel 5.9 on a thinkpad t480s and I can't see a difference with the service enabled or without.

dziq, 2020-12-09:
I have tested it on my x1c6 and we still need to enable throttled fix. Tested kernel  5.9.12 on Void Linux. Without service enabled feq drops to 2.3MHz after hitting CPU temp about 80C. After throttled enabled I have stable 3MHz with CPU temp 95C.

a-vrma, 2020-12-09:
Fair enough, maybe I tested incorrectly. I doubt Lenovo did anything.

closed 2020-12-22 by erpalma