erpalma/throttled #225: lenovo_fix write to unrecognized MSR

AdrianDeAnda, 2020-11-07:
After upgrading the kernel while running the `lenovo_fix.service` I get this issue every few seconds. In the meantime I have disabled the service.

```
[69.128297] msr: Write to unrecognized MSR 0x1a2 by python3
[69.128297] Please report to x86@kernel.org
```

Hardware:

```
Lenovo Thinkpad X1 Carbon 6th Gen
OS: Arch Linux 
Kernel: 5.9.4-zen1-1-zen 
CPU: Intel i7-8550U (8) @ 4.0GHz 
GPU: Intel UHD Graphics 620 
```

januszmk, 2020-11-08:
@AdrianDeAnda https://git.kernel.org/pub/scm/linux/kernel/git/tip/tip.git/commit/?h=x86/misc&id=a7e1f67ed29f0c339e2aa7483d13b085127566ab try adding allow_writes=on parameter

LinuxOnTheDesktop, 2020-11-10:
This issue is a duplication of #215.

AdrianDeAnda, 2020-11-11:
> This issue is a duplicate of #215.

closed 2020-11-11 by AdrianDeAnda