erpalma/throttled #227: Fedora feedback

lhirlimann, 2020-11-13:
I added the copr repro and tried the install nothing was available ( running F33). 

Tried the manual method but things crash as
```
[ludovic@saraan]~/src% sudo ./throttled/install.sh                                       
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Unable to symlink '/usr/bin/python3' to '/opt/lenovo_fix/venv/bin/python3'
Requirement already satisfied: wheel in ./venv/lib/python3.9/site-packages (0.35.1)
Requirement already satisfied: configparser==3.5.0 in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 1)) (3.5.0)
Collecting dbus-python==1.2.8
  Using cached dbus-python-1.2.8.tar.gz (787 kB)
Collecting PyGObject==3.30.0
  Using cached PyGObject-3.30.0.tar.gz (711 kB)
Processing /root/.cache/pip/wheels/41/fe/c0/bf4275a9f9326a7e65da52c9d48488d3ac8c1f5bc4bf83024a/pycairo-1.20.0-cp39-cp39-linux_x86_64.whl
Building wheels for collected packages: dbus-python, PyGObject
  Building wheel for dbus-python (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-su40i3si/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-su40i3si/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-c0mbk7c_
       cwd: /tmp/pip-install-su40i3si/dbus-python/
  Complete output (268 lines):
  running bdist_wheel
  running build
  creating /tmp/pip-install-su40i3si/dbus-python/build
  creating /tmp/pip-install-su40i3si/dbus-python/build/temp.linux-x86_64-3.9
  checking whether make supports nested variables... yes
  checking whether to enable maintainer-specific portions of Makefiles... no
  checking for gcc... gcc
  checking whether the C compiler works... yes
  checking for C compiler default output file name... a.out
  checking for suffix of executables...
  checking whether we are cross compiling... no
  checking for suffix of object files... o
  checking whether we are using the GNU C compiler... yes
  checking whether gcc accepts -g... yes
  checking for gcc option to accept ISO C89... none needed
  checking whether gcc understands -c and -o together... yes
  checking how to run the C preprocessor... gcc -E
  checking for grep that handles long lines and -e... /usr/bin/grep
  checking for egrep... /usr/bin/grep -E
  checking for ANSI C header files... yes
  checking for sys/types.h... yes
  checking for sys/stat.h... yes
  checking for stdlib.h... yes
  checking for string.h... yes
  checking for memory.h... yes
  checking for strings.h... yes
  checking for inttypes.h... yes
  checking for stdint.h... yes
  checking for unistd.h... yes
  checking minix/config.h usability... no
  checking minix/config.h presence... no
  checking for minix/config.h... no
  checking whether it is safe to define __EXTENSIONS__... yes
  checking for a BSD-compatible install... /usr/bin/install -c
  checking whether build environment is sane... yes
  checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
  checking for gawk... gawk
  checking whether make sets $(MAKE)... yes
  checking for style of include used by make... GNU
  checking whether UID '0' is supported by ustar format... yes
  checking whether GID '0' is supported by ustar format... yes
  checking how to create a ustar tar archive... gnutar
  checking dependency style of gcc... gcc3
  checking build system type... x86_64-pc-linux-gnu
  checking host system type... x86_64-pc-linux-gnu
  checking for native Windows host... no
  checking how to print strings... printf
[... 563 lines cut ...]
        setup(
      File "/opt/lenovo_fix/venv/lib64/python3.9/site-packages/setuptools/__init__.py", line 165, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib64/python3.9/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib64/python3.9/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib64/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/opt/lenovo_fix/venv/lib64/python3.9/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib64/python3.9/distutils/command/install.py", line 557, in run
        self.run_command('build')
      File "/usr/lib64/python3.9/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib64/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-su40i3si/dbus-python/setup.py", line 67, in run
        subprocess.check_call(['make', '-C', builddir] + make_args)
      File "/usr/lib64/python3.9/subprocess.py", line 373, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['make', '-C', '/tmp/pip-install-su40i3si/dbus-python/build/temp.linux-x86_64-3.9', 'pythondir=/tmp/pip-install-su40i3si/dbus-python/build/lib.linux-x86_64-3.9', 'pyexecdir=/tmp/pip-install-su40i3si/dbus-python/build/lib.linux-x86_64-3.9']' returned non-zero exit status 2.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /opt/lenovo_fix/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-su40i3si/dbus-python/setup.py'"'"'; __file__='"'"'/tmp/pip-install-su40i3si/dbus-python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-qc9lafsj/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.9/dbus-python Check the logs for full command output.
```

jallentc, 2020-11-29:
I have this same exact issue w/ the COPR repo, and the same errors while installing manually on Fedora 33.  Everything has worked on releases 30-32, but recently broke on 33. The service fails to start with the following error:

`sudo service lenovo_fix status
Redirecting to /bin/systemctl status lenovo_fix.service
Warning: The unit file, source configuration file or drop-ins of lenovo_fix.service changed on disk. Run 'systemctl daemon-reload' to reload units.
● lenovo_fix.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2020-11-29 01:51:58 EST; 1h 8min ago
    Process: 3967 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
   Main PID: 3967 (code=exited, status=1/FAILURE)

Nov 29 01:51:58 system systemd[1]: Started Stop Intel throttling.
Nov 29 01:51:58 system python3[3967]: Traceback (most recent call last):
Nov 29 01:51:58 system python3[3967]:   File "/opt/lenovo_fix/lenovo_fix.py", line 6, in <module>
Nov 29 01:51:58 system python3[3967]:     import dbus
Nov 29 01:51:58 system python3[3967]: ModuleNotFoundError: No module named 'dbus'
Nov 29 01:51:58 system systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 01:51:58 system systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
`

tadd, 2021-01-04:
Almost the same here with Debian testing, but I realized the contradiction between the fix commit d6aeb6851d1bdbf275f351bfa54adc1e809f1209

> dbus-python==1.2.16

and your (and my) error message

> Collecting dbus-python==1.2.8

then I tried to uninstall with removing all files like `rm -rf /opt/lenovo_fix /etc/systemd/system/lenovo_fix.service /etc/lenovo_fix.conf`.

This fixed my situation, now `sudo ./install.sh` worked well. But `rm /opt/lenovo_fix/requirements.txt` would be a better solution IMO.

erpalma, 2021-01-04:
There was a bug in `install.sh`, see ecabdc2. Thanks for pointing this out.