erpalma/throttled #228: Linux 5.9+ limiting MSR writes?

goodboy, 2020-11-13:
I'm pretty sure the kernel team has now limited write access of MSR registers as per:
- https://www.phoronix.com/scan.php?page=news_item&px=Linux-Filter-Tightening-MSRs
- https://github.com/torvalds/linux/search?q=MSR&type=commits

I haven't been able to get this (very very lovely service) script to work since a couple weeks ago.

I am seeing the `dmesg` warnings:
```
[  894.899410] msr: Write to unrecognized MSR 0x150 by python
               Please report to x86@kernel.org
```

I also get the same read errors as in #216.

The part at the end of that article:
>Ultimately they will likely transition to that as the default at a later time to ensure MSR writes from user-space do not happen. This behavior right now can be toggled via the msr.allow_writes= kernel module paramrter with on/off/default. Should legitimate use-cases come up where writes to MSRs from user-space are still desired, they may add the infrastructure to selectively grant/deny access to specific MSRs and ensure they are sanitized by the kernel.

I honestly haven't dug into the commit list enough and am definitely no kernel expert so I'm just putting this up both as a question and a spot where others can come to report in if experiencing the same.

It'd be great to get a recipe going for how to get back to making this all work with newer kernels.

Thanks again to INTC for making all of this possible :smirk_cat:.

mkogan1, 2020-11-15:
@goodboy you need to addd `msr.allow_writes=on` to your kernel parameters and reboot:
afater reboot it should show it as below:
```
cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.8.18-300.fc33.x86_64 ... msr.allow_writes=on
```
the way to do it is depending on the distro, on fedora for example:
`sudo vim /etc/default/grub`
append `msr.allow_writes=on"` to the end of the GRUB_CMDLINE_LINUX="rd.lvm.lv=fedora_localhost-live/root rhgb quiet" line 
and to apply it find the location of your grub.cfg file
`sudo find /boot -name grub.cfg`
and place the location after the `-o` like below (do not just copy-paste)
`sudo grub2-mkconfig -o /boot/efi/EFI/fedora/grub.cfg`
and reboot

goodboy, 2020-11-17:
@mkogan1 agreed, this is what I quoted in the article as the workaround but I'm wondering is there a plan to follow and work with:

> Should legitimate use-cases come up where writes to MSRs from user-space are still desired, they may add the infrastructure to selectively grant/deny access to specific MSRs and ensure they are sanitized by the kernel. 

I'm thinking if the kernel team thinks msr writes isn't the safest thing in the world then maybe projects like this should be mindful of that and try to find the *safest* way to allow users to work around hw manfs terrible firmware and limitations.

antenore, 2020-12-18:
I confirm adding `msr.allow_writes=on` fixes the problem for me.

LinuxOnTheDesktop, 2021-02-17:
Sorry, but does this mean that, for kernels >- 5.9, we need no longer add `msr.allow_writes=on` to the system's boot string?

erpalma, 2021-02-17:
Exactly yes.

LinuxOnTheDesktop, 2021-02-17:
Right. Thanks. So should this issue not be closed?

erpalma, 2021-02-17:
Well, actually that's just a temporary workaround. We still need a proper solution.

lakotamm, 2021-03-29:
Fedora 34 brings new issues with MSR and I do not seem to be able to find a solution. At the moment none of the kernel parameters seems to fixes the issue. 
Kernel  - 5.11.10-300.fc34.x86_64
throttled version - 0.8-3, installed from copr
Secure boot setting in BIOS - Disabled
lsm parameters:
> lockdown,capability,yama,selinux

Systemctl status reported error: 
> Mar 29 13:35:34 fedora throttled[1161]: [E] Unable to write to MSR. Try to disable Secure Boot and check if your kernel does not restrict access to MSR.
> Mar 29 13:35:34 fedora systemd[1]: throttled.service: Main process exited, code=exited, status=1/FAILURE
> Mar 29 13:35:34 fedora systemd[1]: throttled.service: Failed with result 'exit-code'.

Attempted kernel parameters:

> BOOT_IMAGE=(hd0,gpt4)/vmlinuz-5.11.10-300.fc34.x86_64 root=UUID=906d272f-a74d-4844-8ba3-03ea526bb9be ro rootflags=subvol=root rhgb quiet acpi_osi=!Linux-Lenovo-NV-HDMI-Audio lsm=capability,yama,selinux

> BOOT_IMAGE=(hd0,gpt4)/vmlinuz-5.11.10-300.fc34.x86_64 root=UUID=906d272f-a74d-4844-8ba3-03ea526bb9be ro rootflags=subvol=root rhgb quiet acpi_osi=!Linux-Lenovo-NV-HDMI-Audio msr.allow_writes=on

 

I am using Lenovo T490 with both Manjaro Stable 21.0 and Fedora 34 Beta.