erpalma/throttled #229: Fedora 33 permission error on service

timrichardson, 2020-11-14:
I'm getting a permission error on the lenovo_fix.service

It runs fine from the terminal, it's a SE Linux thing i think. 

```
Nov 15 10:15:13 moncrief systemd[1]: Started Stop Intel throttling.
Nov 15 10:15:13 moncrief audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=lenovo_fix comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Nov 15 10:15:13 moncrief audit[4295]: AVC avc:  denied  { read } for  pid=4295 comm="(python3)" name="python3" dev="nvme0n1p1" ino=1703949 scontext=system_u:system_r:init_t:s0 tcontext=unconfined_u:object_r:unlabeled_t:s0 tclass=lnk_file permissive=0
Nov 15 10:15:13 moncrief systemd[4295]: lenovo_fix.service: Failed to execute command: Permission denied
Nov 15 10:15:13 moncrief systemd[4295]: lenovo_fix.service: Failed at step EXEC spawning /opt/lenovo_fix/venv/bin/python3: Permission denied
Nov 15 10:15:13 moncrief systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=203/EXEC
Nov 15 10:15:13 moncrief sudo[4291]: pam_unix(sudo:session): session closed for user root
Nov 15 10:15:13 moncrief audit[4291]: USER_END pid=4291 uid=0 auid=1000 ses=3 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 msg='op=PAM:session_close grantors=pam_keyinit,pam_limits,pam_keyinit,pam_limits,pam_systemd,pam_unix acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Nov 15 10:15:13 moncrief audit[4291]: CRED_DISP pid=4291 uid=0 auid=1000 ses=3 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 msg='op=PAM:setcred grantors=pam_env,pam_fprintd acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Nov 15 10:15:13 moncrief systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
Nov 15 10:15:13 moncrief audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=lenovo_fix comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
```

timrichardson, 2020-11-14:
`sudo setenforce Permissive `
is a workaround, but I don't know if this is a good idea

timrichardson, 2020-11-16:
I am completely new to SE Linux. This shows there is a difference between the lenovo_fix service and other service files. 

```
[tim@moncrief system]$ ls -Z *.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.bluez.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.fedoraproject.FirewallD1.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.freedesktop.Avahi.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.freedesktop.ModemManager1.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.freedesktop.nm-dispatcher.service
    system_u:object_r:systemd_unit_file_t:s0 dbus-org.freedesktop.resolve1.service
    system_u:object_r:systemd_unit_file_t:s0 dbus.service
    system_u:object_r:systemd_unit_file_t:s0 display-manager.service
unconfined_u:object_r:systemd_unit_file_t:s0 lenovo_fix.service
[tim@moncrief system]$ 
```

timrichardson, 2020-11-16:
I fixed that, but there is still an error on the python process. This is way over my head. I set selinux to permissive mode.

timrichardson, 2020-11-16:
This blog has good tips:
http://www.martin-rdz.de/index.php/2019/12/12/selinux-systemd-and-python-virtual-environments/#comment-50

a working policy is: 

```

module custom_rule 1.0;

require {
	type init_t;
	type unlabeled_t;
	class lnk_file read;
}

#============= init_t ==============
allow init_t unlabeled_t:lnk_file read;
```

Obihoernchen, 2020-12-08:
Use the COPR repo to install on Fedora: https://copr.fedorainfracloud.org/coprs/abn/throttled/
It does not have any SELinux issues.

closed 2021-04-09 by timrichardson