erpalma/throttled #232: It would be nice if HWP load threshold could be configurable

DraganBocevski, 2020-12-30:
If I set HWP_Mode to true, the frequency hovers around 3GHz when the CPU is idling. 
When HWP_Mode is false, the frequency hovers around 0.8GHz when idling.

If it can be controlled in software, It would be nice if we can configure the frequency to go up only under more serious load.

erpalma, 2020-12-30:
With HWP_Mode on I'm averaging at 1GHz on idle. Clearly at the slightest load your frequencies will go up, which is what you want. Are you asking to set the exact integer value rather than a on/off switch?

DraganBocevski, 2020-12-30:
Yes .. for example, I want HWP_Mode to kick in only if there is a sustained load. 
Otherwise, having that my system constantly generates small workloads here and there, the frequencies hover above 3GHz.

I'm wondering if hovering at 3GHz instead of 1GHz would waste battery life more?! I should check this, since this is my main concern. Also, setting the HWP in the /etc/throttled.conf AC profile, has an effect when I'm on battery as well. Don't know if this is by design or AC/battery is not detected properly on my T14.

erpalma, 2020-12-30:
HWP should be set back to a more conservative value by TLP when switching to battery

DraganBocevski, 2021-01-10:
Yes, it seems TLP handles battery/power switch correctly for me, so I'll handle HWP there. 
I will close this issue since it seems that the issue in my original question cannot be really fine tuned in software.
Thanks for the feedback.

closed 2021-01-10 by DraganBocevski