erpalma/throttled #236: Thinkpad E550 - service fails to load despite kernel parameters adjusted

beliarsclaw, 2021-01-29:
Hello everyone,

I'm very interested in activating this script on my system as I experience CPU throttling from time to time. I have a ThinkPad E550. Tried on  Kernels (latest Linux-lts / Linux 5.10.11.arch1-1).
`lsm=capability,yama msr.allow_writes=on`

Service fails to start, getting the following error: 
`● lenovo_fix.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Fri 2021-01-29 16:03:31 CET; 2min 16s ago
    Process: 436 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
   Main PID: 436 (code=exited, status=1/FAILURE)

Jan 29 16:03:31 haibar python3[436]:     set_hwp(config.getboolean('AC', 'HWP_Mode', fallback=False))
Jan 29 16:03:31 haibar python3[436]:   File "/opt/lenovo_fix/lenovo_fix.py", line 554, in set_hwp
Jan 29 16:03:31 haibar python3[436]:     cur_val = readmsr('IA32_HWP_REQUEST', cpu=0)
Jan 29 16:03:31 haibar python3[436]:   File "/opt/lenovo_fix/lenovo_fix.py", line 193, in readmsr
Jan 29 16:03:31 haibar python3[436]:     raise e
Jan 29 16:03:31 haibar python3[436]:   File "/opt/lenovo_fix/lenovo_fix.py", line 181, in readmsr
Jan 29 16:03:31 haibar python3[436]:     val = struct.unpack('Q', os.read(f, 8))[0]
Jan 29 16:03:31 haibar python3[436]: OSError: [Errno 5] Input/output error
Jan 29 16:03:31 haibar systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Jan 29 16:03:31 haibar systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
`
Is it this Model not supported or the recent kernels managed to completely block how this script interferes with it's behavior? 

Thanks

erpalma, 2021-01-30:
It's failing while reading the MSR. Can you manually test with `rdmsr`?

beliarsclaw, 2021-01-31:
I will if I get the CPU to throttle again. After Kernel update (5.10-11) it hadn't throttled even after stressing the system to the maximum.

On a side note, how do you use the rdmsr exactly for this purpose?

sajonCC, 2021-03-01:
Hi all, 
I've run in to the same issue as @beliarsclaw with my T480. I'm running ubuntu with 5.4.0-66-generic kernel. Do you have any idea how I can get your nice tool to work? I run in to CPU throttling at least two times a day. I really need to fix it. 

@erpalma can you please explain how to use `rdmsr` to provide you with necessary data.

Thanks

erpalma, 2021-03-02:
> Jan 29 16:03:31 haibar python3[436]: cur_val = readmsr('IA32_HWP_REQUEST', cpu=0)

In this particular case throttled is failing in the IA32_HWP_REQUEST MSR, so:

`rdmsr -a 0x774`

sajonCC, 2021-03-02:
@erpalma thank you.

Just for now, without thermal throttling active, when I'm trying
`rdmsr -a 0x774`

This is an output:
`rdmsr: CPU 0 cannot read MSR 0x00000774`

And this is an output from status check:
```
lenovo_fix$ systemctl status lenovo_fix.service
lenovo_fix.service - Stop Intel throttling
Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Mon 2021-03-01 17:02:09 CET; 24h ago
Process: 1148 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
Main PID: 1148 (code=exited, status=1/FAILURE)

Mar 01 17:02:09 T480-SA python3[1148]:     set_hwp(config.getboolean('AC', 'HWP_Mode', fallback=False))
Mar 01 17:02:09 T480-SA python3[1148]:   File "/opt/lenovo_fix/lenovo_fix.py", line 554, in set_hwp
Mar 01 17:02:09 T480-SA python3[1148]:     cur_val = readmsr('IA32_HWP_REQUEST', cpu=0)
Mar 01 17:02:09 T480-SA python3[1148]:   File "/opt/lenovo_fix/lenovo_fix.py", line 193, in readmsr
Mar 01 17:02:09 T480-SA python3[1148]:     raise e
Mar 01 17:02:09 T480-SA python3[1148]:   File "/opt/lenovo_fix/lenovo_fix.py", line 181, in readmsr
Mar 01 17:02:09 T480-SA python3[1148]:     val = struct.unpack('Q', os.read(f, 8))[0]
Mar 01 17:02:09 T480-SA python3[1148]: OSError: [Errno 5] Input/output error
Mar 01 17:02:09 T480-SA systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Mar 01 17:02:09 T480-SA systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
```

Do you have any idea why? I can try to assistance you with troubleshooting. If there is any other test I can perform to provide you with more details. Please let me know.

sajonCC, 2021-03-02:
And that's it, I run in to thermal problem again and the output of rdmsr command is exactly the same.

erpalma, 2021-03-03:
You can't read that (and possibly other) MSR. It can be due to a bios lock, kernel problem or whatever. Try with a older kernel, if this is not solving your issue I don't think there is much you can do.

erpalma, 2021-03-19:
Can you test if any of these MSRs can be read?

```
MSR_DICT = {
    'MSR_PLATFORM_INFO': 0xCE,
    'MSR_OC_MAILBOX': 0x150,
    'IA32_PERF_STATUS': 0x198,
    'IA32_THERM_STATUS': 0x19C,
    'MSR_TEMPERATURE_TARGET': 0x1A2,
    'MSR_POWER_CTL': 0x1FC,
    'MSR_RAPL_POWER_UNIT': 0x606,
    'MSR_PKG_POWER_LIMIT': 0x610,
    'MSR_INTEL_PKG_ENERGY_STATUS': 0x611,
    'MSR_DRAM_ENERGY_STATUS': 0x619,
    'MSR_PP1_ENERGY_STATUS': 0x641,
    'MSR_CONFIG_TDP_CONTROL': 0x64B,
    'IA32_HWP_REQUEST': 0x774,
}
```

sajonCC, 2021-03-19:
Sure. Just one more remark. I've upgrade my kernel to 5.8.0-45.

rdmsr -a 0xCE
`4043df1011400
4043df1011400
4043df1011400
4043df1011400
4043df1011400
4043df1011400
4043df1011400
4043df1011400`

rdmsr -a 0x150
`18563
18563
18563
18563
18563
18563
18563
18563
`

rdmsr -a 0x198
`178b00000f00
17b300000f00
17b300000f00
17dc00000f00
17a200000f00
177900000f00
17cb00000f00
17a200000f00`

rdmsr -a 0x19C
`883b2802
883b2802
883a2802
883a2802
883b2802
883b2802
883a2802
883a2802`

rdmsr -a 0x1A2
`1e640000
1e640000
1e640000
1e640000
1e640000
1e640000
1e640000
1e640000`

rdmsr -a 0x1FC
`24005f
24005f
24005f
24005f
24005f
24005f
24005f
24005f`

rdmsr -a 0x606
`a0e03
a0e03
a0e03
a0e03
a0e03
a0e03
a0e03
a0e03`

rdmsr -a 0x610
`4280e800dd80c8
4280e800dd80c8
4280e800dd80c8
4280e800dd80c8
4280e800dd80c8
4280e800dd80c8
4280e800dd80c8
4280e800dd80c8`

rdmsr -a 0x611
`2a38429
2a38429
2a38429
2a38429
2a38429
2a38429
2a3849e
2a3849e`

rdmsr -a 0x619
`7af721
7af721
7af721
7af721
7af721
7af721
7af721
7af721`

rdmsr -a 0x641
`623c8
623c8
623c8
623c8
623c8
623c8
623c8
623c8`

rdmsr -a 0x648
`12
12
12
12
12
12
12
12`

rdmsr -a 0x774
`rdmsr: CPU 0 cannot read MSR 0x00000774`

erpalma, 2021-03-22:
Ok I've pushed a small fix for this issue. Now if you disable (comment) `HWP_Mode` in config the `0x774` MSR should no more be read/write. I hope that everything else should still work.

closed 2021-03-22 by erpalma

DrumSlayers, 2021-05-07:
> Ok I've pushed a small fix for this issue. Now if you disable (comment or delete) `HWP_Mode` in config the `0x774` MSR should no more be read/write. I hope that everything else should still work.

Hi, the issue still going by my side
`rdmsr -a 0x774
rdmsr: CPU 0 cannot read MSR 0x00000774
`
`[drumslayer@DrumSlayer-LaptopHP-Manjaro throttled]$ sudo systemctl status lenovo_fix
● lenovo_fix.service - Stop Intel throttling
     Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2021-05-08 00:43:06 CEST; 3s ago
    Process: 28986 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
   Main PID: 28986 (code=exited, status=1/FAILURE)

mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:     main()
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:   File "/opt/lenovo_fix/lenovo_fix.py", line 933, in main
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:     set_hwp(config.getboolean('AC', 'HWP_Mode', fallback=False))
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:   File "/opt/lenovo_fix/lenovo_fix.py", line 630, in set_hwp
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:     cur_val = readmsr('IA32_HWP_REQUEST', cpu=0)
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:   File "/opt/lenovo_fix/lenovo_fix.py", line 263, in readmsr
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]:     fatal('Unable to read to MSR {:x}. Unknown error.'.format(msr))
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro python3[28986]: ValueError: Unknown format code 'x' for object of type 'str'
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
mai 08 00:43:06 DrumSlayer-LaptopHP-Manjaro systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
`

I have the latest version of lenovo_fix, and the HWP_Mode is commented in config (/etc/lenovo_fix.cfg)

What can i do ? Thanks

commit 3f3b4420 (2021-03-22): don't read/write HWP MSR if disabled in config file (fix #236)