erpalma/throttled #239: Dell G3 3590

empereira, 2021-02-15:
hello,

I have been noticing in the logs the following messages:

mce: CPU1: Core temperature above threshold, cpu clock throttled (total events = 2311)
mce: CPU1: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU5: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU3: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU0: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU4: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU7: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU6: Package temperature above threshold, cpu clock throttled (total events = 12705)
mce: CPU2: Package temperature above threshold, cpu clock throttled (total events = 12705)

I did some research and ended up here on your github. I installed it on my Dell G3, but at first, the messages still occur. Maybe it is not compatible with this model? Something I can do?

erpalma, 2021-02-16:
This message just tells you that your CPU is throttling. Even if you increase the temperature limit, at some point it will probably throttle and you will get this message. This is a correct behavior.

closed 2021-02-16 by erpalma

empereira, 2021-02-16:
Ok @erpalma !!! Strange, because this also happens at times when the notebook is processing normally, without any apparent human interference. :)

KanchiSato, 2021-02-16:
@empereira 
As you increase the operating frequency of the CPU, the temperature will rise.
You may want to review the power settings.
Or you may want to check the cooling function as well, since it may not be working properly.

English is not my mother tongue, so I'm sorry if it's hard to read

erpalma, 2021-02-17:
> Ok @erpalma !!! Strange, because this also happens at times when the notebook is processing normally, without any apparent human interference. :)

So you raised `Trip_Temp_C` and still getting these messages while your notebook is idle?