erpalma/throttled #241: Support for Tiger Lake

Oisangui, 2021-03-08:
I am not able to start the daemon because the i7-1185G7 is not supported. I understand the 11th gen processors can't be undervolted, however, it would be good if the non undervolt capabilities were available.

erpalma, 2021-03-09:
I guess you are right. Are you sure that you can change other settings on 11th gen though?

Oisangui, 2021-03-12:
Yes. For example, I am using the Dell Latitude 7420 shipped with Ubuntu 20.04. I dual booted with Windows because I was frustrated with the thermal throttling in Ubuntu which caused the CPU to stick to 60C.

The thing is, in Windows there is no issue at all. My CPU happily goes to a full 100C for quite a lot of time, which is expected if you want the best performance.

Now, if Windows can do it, why couldn't Linux as well?

erpalma, 2021-03-19:
With the latest commit if you omit `[UNDERVOLT]`, `[UNDERVOLT.AC]` and `[UNDERVOLT.BATTERY]` from config file, undervolt won't be applied.

commit 0b219890 (2021-03-19): don't apply undervolt if omitted from config (fix #241)
closed 2021-03-19 by erpalma