erpalma/throttled #247: Donation option

leithhobson, 2021-04-04:
Considering that this script prevents my CPU from running constantly (literally 100% of the time) at 460MHz, and literally holds it over 3.4GHz, it has saved me from having to replace my laptop for over a year now, and thus I (and I am sure others like me) would be happy to have the option to donate at least a cup of coffee to you :)

Also, genuinely, thank you!

erpalma, 2021-04-07:
That's very kind of you, thanks! However, I would prefer if you donate to any scientific research projects (e.g. cancer treatments) or local animal shelters!

rad8329, 2021-04-08:
Thanks, it works like a charm on Thinkpad T14 Gen1, using Fedora 33. Similar to @leithhobson, I almost got to change my laptop, event I've bought a new drive (Samsung 970 EVO plus) to discard any possible bottleneck, that last resort was because I've noticed this problem has ruined the PHPStorm indexing process, some projects could take 8 min, and now with this patch, it is about 80 sec. 

Also, a very very genuinely, thank you!