erpalma/throttled #253: Undervolting not working

alexrocco, 2021-05-09:
Hi,

I'm trying to configure the undervolt but it seems that it's not working.

Lenovo ThinkPad P14s Gen1 with popos 20.10

inxi 
```
CPU: Quad Core Intel Core i7-10510U (-MT MCP-) speed/min/max: 800/400/4900 MHz Kernel: 5.11.0-7614-generic x86_64 Up: 11m 
Mem: 1252.4/31766.2 MiB (3.9%) Storage: 953.87 GiB (1.2% used) Procs: 256 Shell: Bash inxi: 3.1.07 
```
/etc/lenovo_fix.config
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online
# Auto reload config on changes
Autoreload: True

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: True
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0
# Disable BDPROCHOT (EXPERIMENTAL)
Disable_BDPROCHOT: False

# All voltage values are expressed in mV and *MUST* be negative (i.e. undervolt)! 
[UNDERVOLT]
# CPU core voltage offset (mV)
CORE: -40
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: -40
# System Agent voltage offset (mV)
[... 170 lines cut ...]
[D] core 7 thermal status: prochot or forcepr log = 0
[D] core 7 thermal status: crit temp status = 0
[D] core 7 thermal status: crit temp log = 0
[D] core 7 thermal status: thermal threshold1 status = 0
[D] core 7 thermal status: thermal threshold1 log = 0
[D] core 7 thermal status: thermal threshold2 status = 0
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 0
[D] core 7 thermal status: power limit log = 0
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 0
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 64
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0xf - read 0xf - match OK
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x42816000dd80e8 - read 0x42816000dd80e8 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x42816000dd80e8 - read 0x42816000dd80e8 - match OK

```

Thanks,
Alexandre.

joeftiger, 2021-05-09:
I think the undervolt entry should be `[UNDERVOLT.BATTERY]` or `[UNDERVOLT.AC]`. 

At least that's inside my default config on arch.

alexrocco, 2021-05-10:
> I think the undervolt entry should be `[UNDERVOLT.BATTERY]` or `[UNDERVOLT.AC]`.
> 
> At least that's inside my default config on arch.

The README says that [UNDERVOLT] could also be used, and I've also tried that but no luck.

erpalma, 2021-05-10:
It seems like `[UNDERVOLT]` section does not get loaded from the config file. Are you sure that you are editing the correct config file?

alexrocco, 2021-05-10:
I found something, the first loop shows the error

```
[I] Detected CPU architecture: Intel Comet Lake-U
[I] Trying to unlock MSR allow_writes.
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 23
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[D] cpu platform info: feature ppin cap = 0
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 2
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[D] Undervolt plane CORE - write -40 mV (0xfae00000) - read 0 mV (0x0) - match ERR
[D] Undervolt plane GPU - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane CACHE - write -40 mV (0xfae00000) - read 0 mV (0x0) - match ERR
[D] Undervolt plane UNCORE - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] Undervolt plane ANALOGIO - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] HWP - write "0x80" - read "0x80" - match OK
```
So it basically can write to MSR, but the read does not match? Could it be something related to the laptop, like the BIOS or something else?

erpalma, 2021-05-11:
That MSR is probably locked, it's quite common on latest generations.